LaserWeb/LaserWeb4 #186: progress indicator shows incorrect percentage

rilliam, 2017-03-04:
[image]

jorgerobles, 2017-03-04:
@rilliam, any issue of this kind opened without the workspace.json or any other sources to test will be closed, sorry.

jorgerobles, 2017-03-04:
@openhardwarecoza I've seen other github repos show guidelines on text area upon issue opening. Is that possible for ours?

jorgerobles, 2017-03-04:
@rilliam answering to the issue, the percentage is based on jobs. As could not know what chunk of the bar could represent a large raster.

So we have  a preflight, then a job each operation, and some raster, percentage can only be calculated if all the elements are in the same queue, that due it's nature cannot do actually.

@tbfleming need to split preflight/cut workers code postMessage (those infamous lines)

openhardwarecoza, 2017-03-04:
Great idea! I like the stern approach. Users need to learn, open a properly
debuggable issue, and you can have a good response in no time without
wasting the developers time to have to first ask for stuff.

Git issue templates: found this the other day
https://plus.google.com/+PetervanderWalt/posts/JoZnAepXbQG

I'll add us an issue tomorrow

rilliam, 2017-03-04:
I appreciate having requirements for creating a bug report, great idea.

But trying to justifying having a progress bar that has no relationship to the percentage text displayed just makes me laugh. It makes no sense.

openhardwarecoza, 2017-03-04:
Laugh all you want (: - thats work in progress around moving some functions
into webworkers. Instead of laughing, rather contribute: Pull request or
donations are always appreciated much much more than a laugh. Assholes
laugh. Good people help out

rilliam, 2017-03-04:
Hey bud I did donate. But not anymore. I also am a mod on the google plus page.  I also put up other issues that got resolved. Maybe you should look at that before opening your asshole mouth?

rilliam, 2017-03-04:
In fact now I want a refund.

jorgerobles, 2017-03-04:
@openhardwarecoza, @rilliam :dagger: relax :smile: 

@rilliam I'm aware of the issue, but need those files to keep the work.

Please don't flame out. Now children, apology and make peace.

jorgerobles, 2017-03-04:
Really, even all the donations are welcome, the most encouraging thing, at least for me is contribution: knowledge, good vibes, plenty explanation of an issue.

rilliam, 2017-03-04:
lol what a way to treat a donor and supporter. What a joke.

jorgerobles, 2017-03-04:
Just keep calm please.

openhardwarecoza, 2017-03-04:
I'll gladly refund you. Send me a PayPal address

openhardwarecoza, 2017-03-04:
What is your google+ username?

rilliam, 2017-03-04:
Hey Jorge no worries. Its a great piece of software and you are appreciated. 

I agreed with Peter about the guidelines for tickets. Then I was called an asshole and told to donate when I already have. Not cool but life goes on.

I don't agree with the cosmetic relationship happening with the stupid progress bar. Why you would need a json file to see that, I have no idea but rules are rules. Its not even worth pulling the json file. I thought it might help because it wasn't doing that before but now it is. But again, who cares. Its more important to follow some amorphous guidelines for posting a ticket.

Peter: I changed my mind about a refund, you pissed me off. I am over it. If you want to stalk me online and figure out who I am on there go for it dude but I think you should get some therapy if you thats the case.

openhardwarecoza, 2017-03-04:
Remember who started this by laughing at the developers after Jorge did reply saying where the issue is. Its impossible to collate random github usernames to g+ names, to donations unless you use the same name everywhere. So dont expect privilege based on donations. As i said in the same sentence, contribute with a Pull request is also welcome.  Go anywhere in life and laugh at some team's work. You would get the same response. Read

openhardwarecoza, 2017-03-04:
I also never called you an asshole. I said in general assholes laugh. But if the show fits

rilliam, 2017-03-04:
Dude the word "laugh" could mean anything. You want to take it all negative and get butt hurt about it that is your problem. Jorge didn't seem to take it that way. This is online, lots of context is missing.

I literally just laughed to myself and thought hey maybe I don't know what I am talking about and let it go.

Lets be friends. I will make you some spaghetti.

jorgerobles, 2017-03-04:
@rilliam, the reason is that to debug the issue i have to know what portion of code is responsable.  be sure i have done a bunch of tests and the portion drawing on raster are well done. 
But recently we have done some optimization in vector computing, that's my ping to tbfleming. Maybe an interaction of the optimizations did that weird behaviour.

I have made my mind for this cases long ago. Just attach the files and will do my best.

rilliam, 2017-03-04:
Jorge - got it. Thanks.

jorgerobles, 2017-03-04:
@rilliam I've done some optimizations regarding gcode computing. Please test again with the file that did act weird. If keeps behaving wrong, please attach. Thanks

closed 2017-03-04 by jorgerobles