LaserWeb/LaserWeb4 #187: Add zoom to max object

cojarbi, 2017-03-04:
We have reset view but i find that for smaller objects we could use a button to zoom and place the object (s) max to the screen

cprezzi, 2017-03-04:
Also my opinion. Best would be to have multiple view presets like "zoom objects (top)", "zoom objects (isometric)" "workspace (top)"...

openhardwarecoza, 2017-03-22:
Something like this from #155 

[image]

cojarbi, 2017-03-22:
I still would like to see a button to zoom max on what there is on the canvas

tbfleming, 2017-05-21:
I replaced the Reset View button with two zoom buttons: Mach (machine) and Doc. Doc zooms to the selected documents, or to the entire set if none are selected.

jorgerobles, 2017-05-25:
@tbfleming I did a patch (not commited ) about views:

<pre>
 <button onClick={e=> {this.props.dispatch(setCameraAttrs({ eye:[-150,100,0], up:[0,0,1] }))}}>Left</button>
 <button onClick={e=> {this.props.dispatch(setCameraAttrs({ eye:[450,100,0], up:[0,0,1] }))}}>Right</button>
 <button onClick={e=> {this.props.dispatch(setCameraAttrs({ eye:[150,-200,0], up:[0,0,1] }))}}>Bottom</button>
 <button onClick={e=> {this.props.dispatch(setCameraAttrs({ eye:[150,400,0], up:[0,0,1] }))}}>Top</button>
</pre>

Made fast & dirty moving the camera on certain position and capturing values. Thinking on it, maybe we can expose a "named view snapshot" as a workspace setting or so. User moves to their desired camera point and then capture that view, naming it, saving on localstorage or work.

closed 2017-07-08 by cojarbi