LaserWeb/LaserWeb4 #188: Z Stage required fields > info bubble

cojarbi, 2017-03-04:
When enabling z stage there is a start height field added to the operations options. Is this information bubble necessary
[image]
?

jorgerobles, 2017-03-04:
Was an issue opened by @Darklylabs

cojarbi, 2017-03-04:
I dont this this as an open issue. I know @DarklyLabs needs the field but not sure if the bubble is needed, cosmetic but it can be perceived as a missing field.

jorgerobles, 2017-03-04:
Well, was not opened as an issue on github but sent me a mail and done as they requested. I agree that could be better a warning than an error, but that involves some refactoring on bubbles . I can change to any other styling if more people interested. Indeed, the bubbles are not acting as "errors" because they don't prevent operation execution.

cojarbi, 2017-03-04:
Is this the only one thats not required? Most suer wont see it unless you turn on Z stage

jorgerobles, 2017-03-04:
Yes, this is the only one so far. And won't be turned on if don't have Z stage.

DarklyLabs, 2017-03-04:
This is a reminder field for users to set their correct z-height.
It is important to not be missed to avoid ruining a piece of material or causing a collision with material.
I don't see it as a big deal as it doesn't appear unless z-stage is enabled.

tbfleming, 2017-03-05:
This warning should not be there since it only applies to emblaser. Z=0 is the starting value for machines with a top-of-material sensor.

DarklyLabs, 2017-03-05:
@tbfleming Is it really that much of a problem to remind users to set this?
Is it causing any problems?

tbfleming, 2017-03-05:
@DarklyLabs 0 is a valid value for non-emblaser machines; actually it will be the most common value for non-emblaser machines. I don't want to train those users to ignore messages like Microsoft does with all their useless crap in the Windows 10 notification area.

jorgerobles, 2017-03-05:
@tbfleming, @DarklyLabs, @cojarbi.

- A solution is to present an optional Z default starting height (I think this was already discussed)
 This field will be allow to be with a machine profile, and only fills the default value of starting height on all the jobs. If disabled, no fill is done.
- Other way is to change the field to a text field, allowing, so prefilled with empty value. Then the validation will be clearly valid, as no work with Z could be without filing a number in that field.

cprezzi, 2017-03-05:
I would prefer to have this value preset in the machine profile, or at least in the settings.
It's very annoying to see that red flyout on every operation.

tbfleming, 2017-03-05:
@jorgerobles that sounds like a good plan, but do both of your points: change it to a string, add a setting for the default value (will be empty for emblaser), and make it complain if the fields doesn't have a number (positive, negative, and 0 are all valid).

jorgerobles, 2017-03-05:
Done. now we have a machineZStartHeight settings to customize starting operation Z height. Must be a number, or will complain.

closed 2017-03-05 by jorgerobles