LaserWeb/LaserWeb4 #189: Missing module

tbfleming, 2017-03-05:
The following doesn't work on Windows 10 (dev-es6 branch):

```
git pull
npm run installdev
npm start
```

This produces: ```Cannot find module "socket.io-client"``` in the browser console.

cprezzi, 2017-03-05:
@tbfleming This module usualy doesn't have to be added manually, because it's served from every node server by default. I don't know if we get possible version problems, if we add it manually.

tbfleming, 2017-03-05:
Reopening since StackOverflow answers conflict.

Some answers:
1. It serves the file automatically
2. It doesn't serve the file automatically without ```npm -i save```

1 is the highest-upvoted answer, yet that's not what it's doing on my machine. My machine behaves like 2.

tbfleming, 2017-03-05:
I suspect StackOverflow people who accepted 1 ```npm i -g```'d something which included ```socket.io-client```.

cprezzi, 2017-03-06:
Could this also be related to the issue, that lw.comm-server is not updated with npm run installdev?

tbfleming, 2017-03-06:
Possibly, but I'm skeptical that the server would automatically serve socket.io-client without it being installed.

installdev not updating server: moving away from ```git+``` entries in ```package.json``` should help. See https://github.com/LaserWeb/LaserWeb4/issues/190#issuecomment-284232325

openhardwarecoza, 2017-03-06:
> package.json should help. See #190 (comment)

@tbfleming
socket.io server does indeed "serve" the client without needing to install
it seperately

See https://github.com/socketio/socket.io-client#how-to-use :

"A standalone build of socket.io-client is exposed automatically by the
socket.io server as /socket.io/socket.io.js. Alternatively you can serve
the file socket.io.js found in the dist folder."

Note they advise serving it yourself, as a secondary option
Even under Electron it does serve it.

tbfleming, 2017-03-06:
That's probably the problem. Webpack-generated code normally pulls from webpack-dev-server, not the socket.io server.

tbfleming, 2017-03-06:
Bundled webpack-generated code normally pulls from the files in dist/.

openhardwarecoza, 2017-03-06:
In Electron we serve socket.io via node-static - then it includes it as
socket.io server is running.

tbfleming, 2017-03-12:
I just wiped node_modules and did a fresh installdev; everything seems to be OK with socket.io-client in package.json.

commit 70801d5d (2017-03-05): Fix #189: missing module
closed 2017-03-18 by jorgerobles