LaserWeb/LaserWeb4 #190: Moving stable version to Master.

jorgerobles, 2017-03-05:
Hey all @openhardwarecoza, @tbfleming, @lautr3k , @cprezzi 

Should we push stabilized versions to master? I mean, many users are complaining some pushes done to dev-es6, that in fact is a dev-evelopment branch. I don't know if this could be a mess, but maybe we have to offer master branch for stable functionalities.

DarklyLabs, 2017-03-05:
This would definitely make things easier for us to distribute the latest working and stable version to users.
Could this also help resolve the electron exe and app compilation problems we are having?

jorgerobles, 2017-03-05:
I hope so, but this implies some extra work, and we need hands.
We need people that installs the latest versions (so a bit of development background will be great ) on development on pre-master, and perform Q&A. That is not a trivial work, be aware.

DarklyLabs, 2017-03-05:
We are still stuck without being able to offer exe and app versions of the software to our users because of an electron compile problem. The only version we can offer curtrently is 4036 which doesn't include a lot of the bug fixes you have applied as well as the most recent Comms from Claudio.
At the moment we have an amazing piece of software which is stuck on GIT.

openhardwarecoza, 2017-03-05:
I pushed some fixes for the cache issue (issue #15 on lw.comm-server)

DarklyLabs, 2017-03-05:
@openhardwarecoza Great! Will check and report back first thing in the morning.

openhardwarecoza, 2017-03-05:
@jorgerobles I dont think this repo needs a master as we never want users downloading/using this directly.  The stable releases will be made available as precompiled Electron applications?

openhardwarecoza, 2017-03-05:
We may want to tag "releases" in this repo eventually to make appveyor/travis trigger on new releases?

cprezzi, 2017-03-05:
I think it would make sence to have a master branch for releases olny, so dev-es6 don't have to care about braking anything.

jorgerobles, 2017-03-05:
I was thinking the same @cprezzi

tbfleming, 2017-03-05:
We need to lock down the following before release:
* Update git submodules
* Remove ```--remote``` from ```installdev```
* Replace the ```git+``` entries in ```package.json``` with released NPM packages or git submodules (I'm leaning toward git submodules)

This will make it possible to rebuild and branch off the release version without submodule pushes breaking it.