LaserWeb/LaserWeb4 #192: npm start errors on Debian 8 (jessie)

emteeoh, 2017-03-06:
I'm getting these errors when node tries to compile module webpack when I run "npm start" 
I can't promise the problem isn't that I didn't install something... Its a brand new laptop with a clean install. I've already had to re-install nodejs because the debian stable version is too old. I'm sure there are other similar problems I haven't found yet.

ERROR in ./components/about.js
Module not found: Error: Cannot resolve module 'marked' in /home/richard/projects/LaserWeb4/src/components
 @ ./components/about.js 19:14-31

ERROR in ./components/about.js
Module not found: Error: Cannot resolve module 'raw-loader' in /home/richard/projects/LaserWeb4/src/components
 @ ./components/about.js 42:119-173

ERROR in ./reducers/machine-profiles.js
Module not found: Error: Cannot resolve 'file' or 'directory' ../data/lw.machines/machine-profiles.json in /home/richard/projects/LaserWeb4/src/reducers
 @ ./reducers/machine-profiles.js 14:21-73
Child worker:

openhardwarecoza, 2017-03-06:
'npm run installdev'

emteeoh, 2017-03-06:
that ran cleanly, (other than some warnings about optional packages for MacOS) but then I did an npm run got the same errors.

emteeoh, 2017-03-06:
hmm. I ran:
npm cache clean
rm -r node_modules
npm install
npm run installdev
npm start

Everything was very clean right up untill npm start tried to build webpack.  The error message I got was:
WARNING in ./~/raw-loader!./src/data/lw.machines/machine-profiles.json
Module build failed: SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at Object.module.exports (/home/richard/projects/LaserWeb4/node_modules/json-loader/index.js:7:48)
 @ ./src/data/lw.machines ./~/raw-loader!^\.\/.*$

ERROR in ./src/data/lw.machines/machine-profiles.json
Module build failed: SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at Object.module.exports (/home/richard/projects/LaserWeb4/node_modules/json-loader/index.js:7:48)
 @ ./src/reducers/machine-profiles.js 14:21-73

openhardwarecoza, 2017-03-06:
Lw.machines is a git submodule. Not an npm package. Npm run installdev (not
npm install) runs the submodule init/update

emteeoh, 2017-03-06:
I ran machine-profiles.js through a JSON validator (jsonlint.com) and it really does seem to be missing a closing curly brace. adding one gets it through jsonlint. I'm going to see what it does when I add it...

emteeoh, 2017-03-06:
And now it runs!

I'm going to look through the revision history of the machine-profiles.json, see where it went wrong, then I'll close this issue.

emteeoh, 2017-03-06:
fabcreator added a new machine to machine-profiles.json incorrectly. Adding a closing curly brace fixes it syntatically, but I don't know if the file makes actual sense: I don't have a laser cutter handy to try it out.

Anyhow, I made a pull request with the extra curly brace.

closed 2017-03-06 by emteeoh