LaserWeb/LaserWeb4 #193: Ability to watch homing/limit switches state

emteeoh, 2017-03-06:
When setting up a new machine, or restoring an old one, its really important to test the limit switches before you try to do a home. LW doesn't show you the state of the limit switches though, so users need to manually spam M119 (on the smoothie) to get the switch status. If we could put a way to see the limit switch statuses, probably on the jog screen? that would rawk.

(incidentally, what's the difference between an enhancement and a feature request?)

openhardwarecoza, 2017-03-06:
Enhancement makes an existing feature better. 

Monitoring endstops seems like a very odd thing to me. Read the manual and wire things properly. Its not really the place for guesswork

closed 2017-03-06 by openhardwarecoza

emteeoh, 2017-03-06:
Its not something I need all the time, but testing the setup is something that needs doing, and wiring and switches fail too. Also, sometimes there is no manual! Not that switches are complicated. I'd hope you can make a NO switch work without needing too much hand-holding.

cprezzi, 2017-03-07:
I think it doesn't make sense to confuse users with the permanent display of the limit switch state in the jog window. Also, many (diode laser) users don't even have limit switches. That's something we can discribe on the setup documentation page (as it's done once).