LaserWeb/LaserWeb4 #195: Show message to user when file is not supported or broken

cojarbi, 2017-03-07:
Currently when loading a file that is not supported or has issues LW keeps quit and can be confusing. 

Can we display a a message telling the user whats wrong?

Bellow 2 responses shown in dev tools for different cases. One is an AI file and the other is a troublesome dxf
[image]
[image]

closed 2017-03-09 by jorgerobles

cojarbi, 2017-03-09:
@jorgerobles This is working for AI files but not in the case of the dxf that not readable like above. Did you include that in the list of reported files?

jorgerobles, 2017-03-09:
@iceblu3710 could you check for a fix on dxf parser, if not i will add a generic catch.

cojarbi, 2017-03-09:
@iceblu3710  in case you need a file that shows the above error
[rezervirano copy.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/832578/rezervirano.copy.dxf.txt)

iceblu3710, 2017-03-10:
What dxf version is the rezervirano file? It has no tables section which is where the styles and layers are suppose to reside. I currently use that for the colorByLayer inharentence.

I can output an error and tell the user to save in a compatible file format.

I will likely have to branch the DXF parser library and bring it into the new century to make a more robust parser system.

cojarbi, 2017-03-10:
@iceblu3710 this files where created from sketchup with the guitar list plug in. Im not sure why its not recognized in AI either. 

What i would want to achieve is that the user gets a warning when theres a problem with the file or is not compatible.

Exporting from sketchup via regular export in either 2000 or 14 get read perfectly

iceblu3710, 2017-03-10:
PR #201 for my additions.

I added a default byLayer color which I should have had anyways and will log an error when an unknown entity is encountered. File loads correctly now.

iceblu3710, 2017-03-10:
I would merge the PR but I don't know if your suppose to merge or rebase...

jorgerobles, 2017-03-10:
It's merged! Thanks!

cojarbi, 2017-03-10:
@iceblu3710 i found another 2 different error messages, maybe you can take a look. Im not sure whats the difference.

[RTM.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/833865/RTM.svg.txt)
[RTM2.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/833866/RTM2.dxf.txt)
[image]

cojarbi, 2017-03-10:
@iceblu3710 that second error you see above must have something to do with the update from yesterday as im getting that qhen opening the file used in #203

iceblu3710, 2017-03-10:
That first error is in the parser which only handles R12 reliably as there is no fail or check mechinisims.

On my PR I added a log error on unknowen entity. I dont know whay an Action entity is but will dig into it this afternoon.

cojarbi, 2017-03-10:
thanks @iceblu3710 
@jorgerobles if possible lets add first error to the list of items to display an alert to the user. Only if we are not supporting it, it would be good to add that requirement in the message " only R12" for example

iceblu3710, 2017-03-10:
I have reworked the fail mechanism for a more verbose fail. Also included a file type reminder.

Just doing some testing now.

iceblu3710, 2017-03-10:
PR #205 has been updated.

[image]

iceblu3710, 2017-03-10:
@cojarbi Changes should be live

cojarbi, 2017-03-10:
@iceblu3710 RTM2 files is generating multiple warnings for me and still get loaded empty. can we make it no load at all and then show only one warning?
http://recordit.co/7FP2RVlfiA

Sorry about RTM.dxf i loaded the wrong file
[RTM.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/835333/RTM.dxf.txt)

Now im worried that if there is not common denominator for how dxf files are generated we could go into a rabit hole.

Is there a way to just exclude all other types not supported with a single error message?

iceblu3710, 2017-03-11:
Ues it is a large problem I found that the dxf key pair system is overly
complex due to its origins waaaay back in the day. For a "properly"
formatted file to be generated I have found only a few programs. Most
script based generators only output thier relevent skeleton.

Programs like DraftSight and some Linux apps will guess the format and fill
in the blanks but that would be a ton of extra work.

I could wrap my entire dxf lib in a try catch and throw the first error to
exit the block? Not sure if this would mess things up or not as sometimes
you still get enough of the file to be useful.

Another option could be to collect all the messages and build one big alert
at the end of the load process.

cojarbi, 2017-03-11:
Maybe the best for the user would be:
If theres an error show just one message, if there something to load ( as you said there could be something useful in there ) still shoe the message but let the file load, the the user can decide to fix it or use whats there

iceblu3710, 2017-03-11:
I started an error log inside my dxf converter and report the findings before handing the state file over. Tested to work on multiple files loaded at once, each gets its own report.

[image]

jorgerobles, 2017-03-11:
Great, can I suggest an 'errors found loadin files, please check console
log' and piping errors to console? will be an option if many files are
opened.

iceblu3710, 2017-03-11:
@jorgerobles I will dump each to the console as well. Will make group copy paste for error reporting easier as well.

cojarbi, 2017-03-11:
Is it strictly necessary to show all those messages, could be confusing to the user

iceblu3710, 2017-03-11:
The user might only care if the file works or not but when they come asking why it would be nice not to have to tell them to check the console every time.

[image]

cojarbi, 2017-03-11:
@iceblu3710 displaying the message its necesary thats why i suggested it, not sure if displaying block 2 block 3 block 4 etc is necesary unless its telling you how to fix it or what needs fixing

iceblu3710, 2017-03-11:
Do you want a generic message on every error like this:
```
Malformed DXF file, please re-save in R12 ASCII format.
Check console log for more details.
```

Then log to console like:
```
Errors found in RTM2.dxf
Unsupported dxf entity: INSERT:block 2
Unsupported dxf entity: INSERT:block 3
Unsupported dxf entity: INSERT:block 4
Unsupported dxf entity: INSERT:block 5
Unsupported dxf entity: INSERT:block 6
Unsupported dxf entity: INSERT:block 7
```

cojarbi, 2017-03-11:
I think it makes more sense for the user. We already know that to work you need the file as R12 ASCII and will add that to documentation. Maybe you can add a " For help on this look at the docuemntation on preparing compatible DXF files on cncpro.co ( maybe we add the direct address ).

What do you guys think?

jorgerobles, 2017-03-11:
I prefer [that](https://github.com/LaserWeb/LaserWeb4/issues/195#issuecomment-285881386) too, thanks!

iceblu3710, 2017-03-11:
Ok, updated and tested to only log real errors and works with multiple file loading. User popup is:

```
Errors found in RTM2.dxf please re-save in R12 ASCII format.
Check console log and documentation for more details.
```

When we have a page in the wiki I can add a link.

cojarbi, 2017-03-11:
@iceblu3710 you haven't push this correct?

iceblu3710, 2017-03-11:
Haven't issued a PR yet. Is it the way we want?

cojarbi, 2017-03-11:
@iceblu3710 i want to get the link but im unable to add the topic to the cncpro.co

cojarbi, 2017-03-12:
@iceblu3710 here is the link http://cncpro.co/index.php/35-documentation/working-with-files/working-with-dxf

Maybe if possible put "Click here" as the link is very long

iceblu3710, 2017-03-12:
"Click Here" looked weird, I think hotlinking the documentation keyword is pretty obvious though. Opens in a new tab.

[image]

iceblu3710, 2017-03-12:
PR #205

commit f651e7c1 (2017-03-09): Fix #195
PR #208 (2017-03-12): Error display message
- Added a link to the documentation
- Closes #195
commit 0d326cda (2017-03-12): Error display message (#208)

- Added a link to the documentation
- Closes #195