LaserWeb/LaserWeb4 #196: Operations not honoring the order order selected

cojarbi, 2017-03-08:
This was tipped by Walter K on G+

Operation order seems to be all vectors the rasters
Attached workspace is 
Laser Raster
Laser Cut
Laser Raster
Laser Fill path

[CRRVRlaserweb-snapshot.json.txt](https://github.com/LaserWeb/LaserWeb4/files/826365/CRRVRlaserweb-snapshot.json.txt)

jorgerobles, 2017-03-08:
@cojarbi, @openhardwarecoza, @tbfleming .. massive refactor of gcode queue. please check.
@tbfleming lasercut and mill webworkers need a progress status on large jobs.

jorgerobles, 2017-03-08:
@tbfleming preflight is getting errors on maximum callstack on worker at https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/lib/workers/cam-preflight.js#L54
Could you please check to provide a better way of preflight progress?

cojarbi, 2017-03-08:
@jorgerobles I can confirm that the update is respecting the order now. Tx

@openhardwarecoza strange is i get Devtools open at launch with this compile but your nightly does not why?

openhardwarecoza, 2017-03-08:
https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/server.js#L2019
defaults to opening it.  I usually run a check, make sure its fine, then
comment out that line, then run final build.  But lets default it to not
open (: i think we are past that issue now. stand by

openhardwarecoza, 2017-03-08:
https://github.com/LaserWeb/lw.comm-server/commit/3f280bd4a2ac1adaf64e328af086bbcd589343b2

Git pull, then build (:

cojarbi, 2017-03-08:
Ok cool I did that just did not know if you had a magic spell that did that

closed 2017-03-09 by jorgerobles