LaserWeb/LaserWeb4 #197: Machine Profile and Electron Cache

DarklyLabs, 2017-03-08:
Should the machine profile for the Emblaser and Fabkit be appearing in the compiled electron releases?

They do not appear when we run the electron 4.0.42 mac and pc versions, but previously imported machine profiles do appear.

Is this a caching issue and if so, how can it be cleared?

jorgerobles, 2017-03-09:
@DarklyLabs It's done. Base profiles (starts with *) are always refreshed.

closed 2017-03-09 by jorgerobles

DarklyLabs, 2017-03-10:
@jorgerobles The Emblaser and Fabkit profiles are showing up now. Well done.
We still have remanence of perviously saved profiles listed. Where is this stored and how can they be removed?
 Perhaps a complete 'reset to default' option to return everything in LW4 back to a default state?

jorgerobles, 2017-03-10:
It's stored on localstorage. To delete an stored profile, use the delete
button on under profile selecting it first.
I could add a reset profiles under Settings->Tools

DarklyLabs, 2017-03-10:
@jorgerobles A reset profiles would be very useful. We have a blank profile that somehow made its way into the mix which cannot be deleted.
Being able to return to a pure vanilla version of LW could be useful for users who have been experimenting and now want a completely clean setup, since a re-install will not give them this.

commit 37464b7a (2017-03-09): Fix #197