LaserWeb/LaserWeb4 #200: Selected Feed units setting label not being displayed correctly in Jog>Control

cojarbi, 2017-03-09:
[image]

openhardwarecoza, 2017-03-09:
To elaborate, it doesnt follow what you selected in Settings tab?

cojarbi, 2017-03-09:
Correct, so if i choose mm/min in settings, jog>control keeps displaying mm/s 
( Am i that bad at describing :)  )

cprezzi, 2017-03-10:
You are right, it should change corresponding to the units set. I probably just forgot it. ;)
I will check that.

cprezzi, 2017-03-10:
Corrected with commit https://github.com/LaserWeb/LaserWeb4/commit/5bbbf72cc8440abdce1c8df1c68107e7f0425bc5

closed 2017-03-10 by cprezzi

cojarbi, 2017-03-12:
@cprezzi not sure if you also want to change this so its consistent
[image]

DarklyLabs, 2017-03-12:
Can anyone explain why the G0 Feedrate is needed here and how its used?
Does it override the default value?
Is it used in GCode generation?

cprezzi, 2017-03-13:
@cojarbi Where is that field from?

DarklyLabs, 2017-03-13:
@cprezzi I think it's the one under the simulator slider

cojarbi, 2017-03-13:
Sorry main screen, bellow the canvas

cprezzi, 2017-03-13:
Ah ok. I did not program that. Think thats for the CAM.

DarklyLabs, 2017-03-13:
@tbfleming Can you shed any light on the reason and function of this slider?

tbfleming, 2017-03-13:
Everything in this area affects the workspace, and nothing else:

[image]

The G0 Feedrate field is for time estimation that the Simulator slider uses.

DarklyLabs, 2017-03-13:
@tbfleming Where is this estimation found?

This naming is ambiguous and easily confused by users as to what it relates to. I can see it useful if you could playback the simulator, but i don't this this is possible?
Either way, a more appropriate name should be used.

tbfleming, 2017-03-13:
The slider position is proportional to simulation time. Far left is time 0 and far right is estimated end time.

The naming is exactly what it says: G0 Feedrate tells the simulator what feed rate G0 is. The simulator doesn't know since it doesn't talk to the firmware.

You can playback the simulator; just move the slider. After you click on it you can also hold down the left or right arrow keys.

DarklyLabs, 2017-03-13:
By calling it G0 Feedrate causes it to be confused with actual G0 feed-rates. As I said, it is ambiguous.

Can I suggest calling it something more appropriate to the simulator?
'Simulator Speed"
"Simulator Estimation Speed"
"Simulation G0"

tbfleming, 2017-03-13:
It's now "Sim G0 Feed".

DarklyLabs, 2017-03-13:
Thx Todd.