LaserWeb/LaserWeb4 #202: Add drag and drop files into the app

cojarbi, 2017-03-10:
Drag and drop files from the explorer / finder into the app as an additional way of loading.
We have this in LW3. Can it be added here?

openhardwarecoza, 2017-03-22:
I think that may confuse users (drag from windows into document tree, then drag from document tree to operations.... Users are going to confuse the two and draf from window to operation i bet.

cojarbi, 2017-03-22:
You might be correct, enough issues we have now but this could be solved by just illuminating the section you are suppose to drag the file to. If it's from outside just illuminate the import section and if it's from LW just illuminate the operations ? 

If to complicated then focus on something more important. I just want to have UI common to now days standards. Most use drag and drop

closed 2017-04-17 by openhardwarecoza