LaserWeb/LaserWeb4 #203: What its supported on the dxf parser?

cojarbi, 2017-03-10:
@iceblu3710 with the purpose to provide answer to our users i would like to know what is and whats not supported at the moment.

Attached you will see the original file and 2 screenshots. This file was exported from AI CS 2017 to DXF R14. 

Base on this we can:
- Read colors strokes
- Read text ( besides the path )
- Read some letters when the text is expanded and ungrouped
- Read layers

It will not 
- Read fill

Is this it or is there something wrong / missing from the above statements

[Colors.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/832968/Colors.dxf.txt)
[DXF LW read.txt](https://github.com/LaserWeb/LaserWeb4/files/832969/DXF.LW.read.txt)
[DXF ORIGINAL AI.txt](https://github.com/LaserWeb/LaserWeb4/files/832970/DXF.ORIGINAL.AI.txt)

iceblu3710, 2017-03-10:
- Read fill (colors.dxf)

    - All fill is disabled as it causes selection issues when drawing larger object over smaller objects. There is no way to figure out the drawing layering in a reasonable way so disabling was the best option.

- Circles & Ovals (shapes.dxf)

    - Your curves are SPLINE entities and not supported, a bulge gives the same result and is supported.

- Read some letters when the text is expanded and ungrouped (colors.dxf

    - Your letters are built with a HATCH entity, not supported.

I am waiting to hear back from the dxf-parser publisher on if he plans on updating the file support. If not I may need to fork the project and make a lw.dxf-parser.

iceblu3710, 2017-03-10:
R12 files on the top, your originals on the bottom:  

[image]

cojarbi, 2017-03-10:
@iceblu3710 thanks, will review once i get home. 

Read Fill- is this selection issue because the user need to click to click on the object or is something else? Could this be solved by using different layers? Because user can manage the paths by the tree

How **exactly** are you producing or fixing the letters and circles. Not all apps use or have the same tools so i need to find a similar one in AI / Inkscape / others.

I just want to be able to get the same result as you and be able to provide that  to the users.
Thanks

cojarbi, 2017-03-11:
Just an update: exporting directly to dxf ( R12 to 2013 ) from Sketchup Pro 2017 are perfectly read by LW4.
importing colors.dxf into Sketchup and then exporting to dxf also produces perfect results

[image]
Left Colors.dxf / Right native draw in Sketchup

iceblu3710, 2017-03-11:
> Read Fill- is this selection issue because the user need to click to click on the object or is something else? Could this be solved by using different layers? Because user can manage the paths by the tree

With fill you can click anywhere in that object to select, so when you want to select say the pocket or cutout on that part you can't. If you had to search the tree manually that would suck with large files. You could design the file with "inside, outside, raster" layers but that only works for files you create. It was decided that the fill didn't convey any information so it was pointless form a cam perspective.

> How exactly are you producing or fixing the letters and circles. Not all apps use or have the same tools so i need to find a similar one in AI / Inkscape / others.

I use the free version of DraftSight as its guaranteed to support everything and can export to everything dxf/dwg related. I also use it at work. I was simply opening your files and save-as R12 ASCII DXF.

> I just want to be able to get the same result as you and be able to provide that to the users.

Would be nice if everyone stuck to the full standards but an empty R12 DXF with no content is 1043 lines long...

cojarbi, 2017-03-11:
Could we use something like this in any way to get perfect files no matter what
https://cloudconvert.com/api

cojarbi, 2017-03-11:
Maybe also this http://www.pstoedit.net/ could get as to read other format? Not sure just a thought

jorgerobles, 2017-03-11:
@cojarbi cloudconvert seems a nice tool. We can move integration to next release (4.1 @openhardwarecoza? ) and it's own issue.

Keeping this issue on topic, just write docs at cncpro.co and this issue could be closed.

iceblu3710, 2017-03-11:
I will wrap all my "optional" sections in try/catch blocks and log that that defaults are being substituted. Relying on a web API would be a major hassle.

We can look at adding cloudconvert into the wiki for problem files though.

jorgerobles, 2017-03-14:
@cojarbi wondering about https://cloudconvert.com/pricing .. Api key should be owned by the user, We only should do a nice use documentation in LW

cojarbi, 2017-03-14:
i was looking for a solution to both svg and dfx issues we have.  This does not solve that

closed 2017-03-14 by cojarbi