LaserWeb/LaserWeb4 #204: installdev broken

cprezzi, 2017-03-10:
Development environment is broken because of some missing libraries in package.json!

[image]

cprezzi, 2017-03-10:
@jorgerobles Could you check please.

jorgerobles, 2017-03-10:
@cprezzi Weird, cannot reproduce the error. Will try a dry run

iceblu3710, 2017-03-11:
I have had problems with the `npm run-script installdev` option downloading to the node_modules and then when the script ends they are not resolvable. running `npm install` works every time.

Only happened a few times on Linux for me, I used nvm to install the latest and everything works fine now.

jorgerobles, 2017-03-11:
@cprezzi still errors?

tbfleming, 2017-03-12:
> I have had problems with the npm run-script installdev option downloading to the node_modules and then when the script ends they are not resolvable. running npm install works every time.

Very odd since installdev runs npm install.

I just wiped node_modules and did a fresh installdev; everything seems to be OK (Windows 10, Node 4.6.0, npm 2.15.9).

iceblu3710, 2017-03-12:
I wiped and re-setup my repo this morning and it worked for me on Linux_x64.
Node v6.9.2
npm v3.10.9

I had my problems back in Feb where installdev said it worked but on npm run it would give package errors. Everything worked well since then.

cprezzi, 2017-03-14:
@jorgerobles The same update (installdev) worked on my notebook, so probably a version conflict on my PC. Will try reinstall.

closed 2017-03-14 by cprezzi

jorgerobles, 2017-03-14:
:)