LaserWeb/LaserWeb4 #207: Add G0 and G1 Filters to View

megashub, 2017-03-12:
I would like to be able to review G0 and G1 paths separately. Right now, they are presented together, so I can't always see exactly where the G0 paths are because they often times blend into the G1s.

jorgerobles, 2017-03-21:
It's done at Webgl level, not to be implemented soon, sorry :|

openhardwarecoza, 2017-03-21:
Are we sure we need this? There are many legit cases where g0 moves may run over g1? I dont see the use of a filter.

jorgerobles, 2017-03-21:
Hmm. the only thing I'm thinking is that the G0 rendering should be moved on top of G1, maybe will suffice? @tbfleming

tbfleming, 2017-03-21:
I'd be tempted to do the opposite: G1 on top of G0.

megashub, 2017-03-21:
Sounds like we're getting into personal preference territory. Perhaps a settings toggle to allow for either one?

--------------------------------------------

 

 
 it on GitHub, or mute
 the thread.

tbfleming, 2017-03-21:
No, that contributes to a flood of settings that most people don't care about. Mozilla fell to that flood and Firefox stripped options out when it rose from the ashes.

cprezzi, 2017-03-22:
@tbfleming @jorgerobles What if you make all G moves only 70% opaque? This way, colors would mix when one move is over the other.

tbfleming, 2017-03-22:
That might work. It will need testing since that could hurt visibility on hi-res displays.

jorgerobles, 2017-03-22:
@megashub in order to go further with this issue, we need a workspace.json to replicate it.
I've used 70% locally but cannot tell If it's enough or not.

megashub, 2017-03-22:
I have only just begun to play around with LW4, so I haven't made many changes other than the very basics to get it working with my K40. Still, if it's valuable, I'll upload my copy of it. I'm also more than happy to just pull down a test build and give it a whirl. Whatever's most helpful to you and the team.

--------------------------------------------

jorgerobles, 2017-03-22:
@megashub, I'm not sure what you mean, but workspace is just a .json file, you can save it from the cam pane top, workpace, save. 

Please just open the file you used to detect the issue, put some operations and generate gcode. Then download the workspace.json, rename as .txt adnd upload here,

Thanks!

openhardwarecoza, 2017-04-17:
Closing for now - for most of the users the green/red suffices

closed 2017-04-17 by openhardwarecoza