LaserWeb/LaserWeb4 #209: What is supported on the SVG Parser?

cojarbi, 2017-03-12:
Today a saw a svg file that was so mixup that i had to almost do it completely new for LW to read it.
I know the user must be able to create the file correctly ( many dont know how ) but i would like to add the documentation on what LW4 expects and what can we accept.

What parser are we using? Can we add alert messages as we did with dxf?

tbfleming, 2017-03-12:
@lautr3k

jorgerobles, 2017-03-21:
@cojarbi I think moving this issue to @lautr3k [repo](https://github.com/lautr3k/lw.svg-parser/issues)
anyway, I could pipe unsupported warnings on dev console to LW console.

cojarbi, 2017-03-21:
Ok

jorgerobles, 2017-03-21:
Ok, for piping for moving for both, ok to get a beer and a pizza? :D

cojarbi, 2017-03-21:
Beer and pizza of course. I know nothing of the rest :). Adding warnings as we did with dxf would be nice to maintain uniformity

closed 2017-04-17 by openhardwarecoza