LaserWeb/LaserWeb4 #210: Small cut path will not generate accurate gcode

cojarbi, 2017-03-12:
@tbfleming  
If the path is to small then the laser cut operation output gcode will not accurate.
It misses or does not follow the path correctly

[image]

tbfleming, 2017-03-12:
The 3 Laser Cut operations now have a Segment field. It defaults to 0mm. Its old non-configurable value was 0.5mm.

Old behavior: Smoothieware was OK with 0.5mm at high feeds. This was too course for some jobs such as this.

New behavior: By default it will reproduce fine detail, but Smoothieware will now choke on some jobs that it could handle before. When this happens, bump up the Segment value. This will cause it to lose detail.

cojarbi, 2017-03-12:
Work beautifully, now i need to add this to the documentation.
I know i had to use the segment on my mill because my gshield was choked on very small segments also.

closed 2017-03-12 by cojarbi