LaserWeb/LaserWeb4 #211: Raster rotation on svg not being read properly

cojarbi, 2017-03-13:
Reported by Chuck on G+

Files created in inkscape 0.91  and the only difference is that the raster is rotate clockwise by 90 degrees.

Opening the files in any other editor will show correctly but in LW4 it shows it mirrored on the Y.

Comparing the  2 files shows:

Test 2 has this while test 1 does not
<image
       **transform="matrix(0,1,-1,0,0,0)"**
       
Also

Test1                                             Test2
id="image4494"
       x="67.640472"		       x="244.61024"
       y="244.61024"		       y="-421.39047"

[image]
[test2.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/838143/test2.svg.txt)
[test1.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/838144/test1.svg.txt)

Same thing happens when created in Illustrator CS2017

tbfleming, 2017-03-13:
Related to #147.

cojarbi, 2017-03-13:
@tbfleming really? i thought 147 was about getting LW to rotate images. This is not the same ( or so i think ). Images rotation are not read properly

tbfleming, 2017-03-13:
It's the same issue under the hood. The current code doesn't understand image rotation, whether that rotation happened outside the app or the user wants to rotate inside the app.

cojarbi, 2017-03-13:
ok understood

closed 2017-03-13 by cojarbi

tbfleming, 2017-03-13:
I do agree that there is a difference from the end user perspective.