LaserWeb/LaserWeb4 #213: Load gcode button unreliable

tbfleming, 2017-03-13:
Sometimes it works and other times it fails to dispatch the GCODE_SET action.

[grayscaletest.gcode.txt](https://github.com/LaserWeb/LaserWeb4/files/838876/grayscaletest.gcode.txt)

jorgerobles, 2017-03-13:
I've tested 20 times, failed 3, but no error is thrown :slightly_frowning_face:

tbfleming, 2017-03-13:
It's a heisenbug :(

jorgerobles, 2017-03-13:
I've applied this workaround but fails equally http://stackoverflow.com/questions/12030686/html-input-file-selection-event-not-firing-upon-selecting-the-same-file

jorgerobles, 2017-03-14:
Well, after new tests, fix does not work (former commit is wrong, that code is not used). It's a native event bug. Sooo, I will close, sorry.

tbfleming, 2017-03-14:
Created a new label for cases like this.

tbfleming, 2017-03-14:
I think we should keep this open so we can point people to it. Also so we'll remember to recheck at the next Chrome release.

jorgerobles, 2017-03-14:
Right, thanks!

tbfleming, 2017-03-14:
The two oldest open bugs in jscut have tags "Safari Bug" and "IE Bug" :)

donkjr, 2017-03-17:
FYI: I noticed a condition under which I can reproduce this error. 
The first time you use "Generate" after loading an image into operations with the "Generate" button in the yellow state and nothing in operations.
The next time you "Generate" it will work correctly and from then on ....

jorgerobles, 2017-03-17:
Will try, I'm not sure. I had randomly misfires (~20%) just trying to load the same file twenty times a row.

jorgerobles, 2017-03-24:
@tbfleming Found. Virtual file inputs are the culprits. Searching a workaround

commit 83df2c20 (2017-03-24): Fix #213
closed 2017-03-24 by jorgerobles