LaserWeb/LaserWeb4 #214: Hiding irrelevant operations / tabs

openhardwarecoza, 2017-03-14:
Example 1:
Use drags only RASTER / IMAGE into operations:

Current Behaviour:  User can select Laser Cut and CNC operations (If not educated, it seems perfectly logical) - but gets NO gcode (since you cant CUT a Raster...)  

Expected Behaviour: Only show operations relevant to IMAGE types.  

Example 2: 
User drags vectors into operation

Current Behavior: User selects Laser Raster because we earlier told him thats the one that "colors in" an image

Expected Behaviour:  Hide Raster operations

Example 3:
User drags Vectors and Bitmaps into operation

Expected Behavior = current behaviour - it really should show both.  But we need to add some documentation to cncpro.co

Example 4:  (Related to #199's plans also) 
User runs a Laser only

Current Behavior:  Shows CNC jobs too

Expected Behavior:  Like we had in LW3:  In Settings: Machine - specify if its a laser, 3d printer or CNC:   If Laser is selected, only Laser type CAM operations are shown.  If CNC, then only CNC operations are shown.  If 3D Printing, we should hide the CAM tab, and add a new tab, 3D Printing that only gets shown in 3DP mode.

jorgerobles, 2017-03-14:
Got it working, but do we traverse all the documents children (snail slow), not?

openhardwarecoza, 2017-03-14:
Not sure (:

I dont know how deep down users will place such craziness.  Just file on its own is probably definite.   But sometimes images can be embedded in an SVG.   How deep?  Inside a group, inside a layer? I dunno (; I'm a dxf guy

jorgerobles, 2017-03-14:
Yes, SVG can embed images :)

openhardwarecoza, 2017-03-14:
So, i take it that means traverse all children? 
Or at least "all children in what was just dropped"

tbfleming, 2017-03-14:
Example 3:
* User drags an entire SVG file into operations. Selects Laser Raster.
* User drags an entire SVG file into operations again. Selects Laser Cut.
Current behavior: operations filter what they support. Why give an error or warn?

openhardwarecoza, 2017-03-14:
@tbfleming true, that can happen too...  Thats why i suggested asking user if you can split it on the first drag maybe?  (trailing G_+ and feedback from the 500+-newbie LW4 users (According to google analytics) most dont understand it.  So, either that, or a documentation fix

tbfleming, 2017-03-14:
I suspect restricting the set of operations based on what's present in the selection may be enough. If there's an image somewhere in the selection, allow raster. If there's a vector somewhere in the selection, allow vector ops.

openhardwarecoza, 2017-03-14:
Agreed... If both are present, presenting all options is the logical answer

jorgerobles, 2017-03-14:
Ok, then traverse children for images. I'm still worried for large files performance.

openhardwarecoza, 2017-03-14:
Ah Performance, that incompatible thing with "user ability"

openhardwarecoza, 2017-03-14:
It amazes me the little things we now bitch about when you compare what we have with what we had before you awesome people joined the project....

openhardwarecoza, 2017-03-14:
When I was working on the CAM tree concept in https://github.com/LaserWeb/LaserWeb3/issues/52 - arthur wolf commented about how I should keep the UI simple...   I made a joke about "note to self, laserweb4" and then a mass of buttons confusing Homer...   I wonder if we did achieve that goal (; 

[image]

jorgerobles, 2017-03-14:
Last thing. What do you prefer, hide operation types or make them unselectable?

cojarbi, 2017-03-14:
Indeed this will solve many many support questions

jorgerobles, 2017-03-14:
Tada!

closed 2017-03-14 by jorgerobles

cojarbi, 2017-03-14:
Wonderfull @jorgerobles

openhardwarecoza, 2017-03-14:
@jorgerobles ! Man that was too fast!

cprezzi, 2017-03-15:
Am I the only one who could only do "laser cut" operation now? When I drag a raster image, the operatino is showing "Laser Raser" but all params are from "Laser cut", so I can't define all params and no gcode is generated.

jorgerobles, 2017-03-15:
Checking...

cprezzi, 2017-03-15:
All vector operations are working, just raster is not. I think you need to fire the dropdown change event after activate/inactivate dropdown items.

jorgerobles, 2017-03-15:
Yes.. that is. Will be easy.

jorgerobles, 2017-03-15:
@cprezzi Done.

cprezzi, 2017-03-15:
Jep, works just fine :)

commit 2180568e (2017-03-14): Fix #214