LaserWeb/LaserWeb4 #216: Workspace performance

jorgerobles, 2017-03-15:
@tbfleming Working on the same old svg, I've realized, Workspace::setCanvas::draw is always rendering even there's no camera movement, or any other workspace activity (no thick-lines active)

Do we have a way to cache the rendering (as shouldComponentUpdate, but for the canvas)?
Maybe we should go framebuffering and do selective updates? (ex.. thick-lines)

tbfleming, 2017-03-15:
Yes, we should.

jorgerobles, 2017-05-26:
@tbfleming Did a test of monkeypatch cache, not commited, that only updates workspace if sync with `shouldComponentUpdate`.

- (+) The performance is maximum
- (-) Disables fancy select animation 

I tried to do with framebuffers but are beyond my reach :)

<pre>
let draw = () => {
            if (!this.canvas)
                return;

            if (this.__updating) {
                this.__updating=false;
            } else {
                return requestAnimationFrame(draw);
            }
</pre>

and

<pre>
  componentDidUpdate(){
        this.__updating=true;
    }
</pre>

domenic-d, 2017-06-07:
We have just been testing the recent 'Show Raster Preview' feature. 
There is a lot of delay while entering values since calculations are done as each digit is entered. For instance If we wanted to set the contrast to 100, calculations are being performed for 1, 10 and 100.

There is really no need to have such a level of interactivity for these types of adjustments.

It would be far less frustrating to process the value after 'enter' has been pressed.

jorgerobles, 2017-06-11:
@tbfleming How difficult is to layer cake display into different framebuffers? 
The monkeypatch I did on display cache is invalidated for example on job run. And that's just the thing I wish to avoid, so no vector re-rendering interferes (clogs cpu) with job and feedback interaction.

tbfleming, 2017-06-11:
Not difficult.

tbfleming, 2017-06-11:
@jorgerobles It now uses frame buffers to cache documents, document selection, and gcode preview. It doesn't cache laser preview yet.

jorgerobles, 2017-06-11:
I bow to you, fine sir!

jorgerobles, 2017-06-11:
So, the "Display Cache" setting I did could be dropped?

tbfleming, 2017-06-11:
Probably

closed 2017-07-10 by jorgerobles