LaserWeb/LaserWeb4 #217: CAM handling Compound Paths

DarklyLabs, 2017-03-15:
We have received a number of support requests from users reporting svg files take a very long time to add as a document.
We have tracked down the issue to compound paths being present in their svg file.
Although it was easy for us to open the svg file in Illustrator and release the compound paths, this is not always possible for users and an extra step for them.
I have attached a typical example. The 'original' file takes 2 minutes top open on our system. The modified file (with no compound paths' takes 2 seconds to open.

[mask2_original.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/845886/mask2_original.svg.txt)
[mask2_noCompoundPaths.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/845887/mask2_noCompoundPaths.svg.txt)

tbfleming, 2017-03-15:
It's spending its time in ```Clipper.SimplifyPolygons```. More evidence that we need to dump handling polygons in javascript and switch it to C++.

openhardwarecoza, 2017-03-22:
Closing this issue for now - the webasm / c++ thing will be a pretty major project, as best 'someday'

closed 2017-03-22 by openhardwarecoza