LaserWeb/LaserWeb4 #218: Image file structured under a layer from Inkscape

donkjr, 2017-03-15:
In Inkscape I created a layer for images and put a bit mapped logo on that layer. Nothing else.
Then I brought it into LW4:
If I select and assign the entire layer to an operation it works fine.
However;
If I select the image only from within that layer and "create single", it will let me define it as that operation but it will not generate the gcode or simulate.

Also: if you define in 'create single"  an image as a "Laser Cut" operation
 it does not prevent the selection but does not generate g-code or allow simulation. 

[image]

openhardwarecoza, 2017-03-16:
Gh-pages isn't as up to date as dev-es6 - the screenshot cut off the address bar, cant see where you are running from? Try a Binary from github.com/LaserWeb/LaserWeb4-Binaries

jorgerobles, 2017-03-16:
@donkjr, the specs of document selection (at #214 ) are these (should add to docs)

1. Must have selected a document to create single/multiple **(just fixed it)**
2. If an operation has only vector documents (traversing each document children), then the raster operations are disabled.
3. If an operation has only raster documents (traversing each document children), then vector operations are disabled.
4. if an operation has mixed documents (several distinct document natures, or a single document whose tree has mixed natures) all the operations are allowed

PS 4,LW has no way to apply different operations to a single tree, so if Laser Cut is selected on a mixed operation that has images, images will not be processed. Either way, if Laser raster is selected on a mixed operation that has vectors, vectors will be ignored.

jorgerobles, 2017-03-16:
@donkjr I've excluded layers (SVG::G) from vector count, so now if dropping a layer that only contains raster, only raster should be allowed.

tbfleming, 2017-03-16:
@jorgerobles instead of looking at the type field, which will gain new values over time, it's easier to look if rawPaths (vector content) or dataURL (image content) aren't empty.

jorgerobles, 2017-03-16:
Neat! Will change :)

jorgerobles, 2017-03-16:
@donkjr please check to close, thanks :)

closed 2017-03-17 by jorgerobles

donkjr, 2017-03-17:
Sorry I was late the the response.
It looks like the issue is indeed improved as I tried the top of the image tree and the image itself and it worked correctly.

THANK YOU!