LaserWeb/LaserWeb4 #219: Add lw.Rasterizer.js's overscan option

tbfleming, 2017-03-16:
I was hoping we wouldn't need it, but it looks like we do. Multiple people have reported edge burning issues.

jorgerobles, 2017-03-16:
Its an existing option?

tbfleming, 2017-03-16:
It appears on https://lautr3k.github.io/lw.rasterizer/

tbfleming, 2017-03-16:
I'm uneasy with the number of options we expose for rasterizing, but it looks like every one is needed.

jorgerobles, 2017-03-16:
Easy way, we (community) elaborate some  operation presets. Common user
only have to touch a couple of params, maybe hiding the rest under advanced
toggle?

tbfleming, 2017-03-16:
That sounds good. Maybe have it remember the toggle as a hidden setting.

cprezzi, 2017-03-16:
I thinks it would be ok to always do overscan. I don't see any reason for switching it off.
Probably the distance of the overscan could be proportional to speed?

tbfleming, 2017-03-16:
I'd rather explain overscan distance to new users than the machine-dependent distance/speed ratio.

jorgerobles, 2017-03-16:
@tbfleming Done. Maybe we should close and move UI operation mods to other issue?

tbfleming, 2017-03-16:
Good idea.

closed 2017-03-16 by jorgerobles

cprezzi, 2017-03-16:
I did mean to not tell the user anything about overscan and have it active all the time. Just internally calculate the overscan distance from speed (hardcoded).

jorgerobles, 2017-03-16:
Both ways: We can use a speed to scan algorithm and put as a placeholder on overscan field if empty. If not empty, entered value overrides. Yay or nay?

openhardwarecoza, 2017-03-16:
Question: how do you overrun when first pixels are at 0 or max edge? Will
we need to introduce homing offsets to have some negative space for the
runup?

jorgerobles, 2017-03-16:
We can warn using red balloons if is on negative space, 
or leave overscan to users who knows what it does and fill with 0, like now (no speed to scan offset :|)

cprezzi, 2017-03-16:
@jorgerobles I just thought we don't even need to show the user an additional field (that we also have to explain), but your idea is also ok.

@openhardwarecoza That's a very good question. Even worse: The CAM doesn't know where the user will set the work offset. So if we do check and warn, we need to do that also after each "set zero".

KirAsh4, 2017-03-16:
Thanks for considering and adding the overscan feature. Can't wait for the build to try it out.