LaserWeb/LaserWeb4 #220: Labels and ID's in Inkscape

donkjr, 2017-03-16:
As a laserweb user I would like it if Laserweb retained and included  inkscapes descriptive properties:

- Layer: Name, 
- Object properties: Label

LW currently keeps the Object properties: ID.
I want to keep the ID as inkscape creates it and identify the object by its Object properties: label.

tbfleming, 2017-03-16:
So far we've been shying away from supporting Inkscape-specific extensions. It opens a rats-nest of supporting the same thing from Corel, Illustrator, etc.

jorgerobles, 2017-03-17:
I agree, we have to narrow to SVG specs, not any external software.
Right now, here's the attributes prone to be used https://www.w3.org/TR/SVG/attindex.html, and only "id" is a real candidate, sorry.
The only option I think is if community develop XSLT (XML transformation) sheets to filter SVG.

donkjr, 2017-03-18:
I agree that if this attribute is not a part of the SVG specs it opens a Pandora's box of compatibility issues.

closed 2017-03-20 by jorgerobles