LaserWeb/LaserWeb4 #222: Webcam: Transformation Filters (v4.2.x targetted release)

openhardwarecoza, 2017-03-18:
Advanced functionality to extend #30 

#191 Note:  Moving this advanced task to its own issue - will be addressed here, when the time is ripe!

openhardwarecoza, 2017-04-17:
@raabrp is working on this...

raabrp, 2017-04-18:
Missed this comment when it was posted, sorry. I'll be sure to post as this goes forward. I'm approaching this somewhat casually, so I hope I'm not holding up anyone else.

openhardwarecoza, 2017-07-31:
Closing this issue for now - months since anyone thought of it

jorgerobles, 2017-07-31:
Just keep it open during summer :)

jorgerobles, 2017-08-03:
@tbfleming I'm taking a look to the code DarklyLabs made to backport to the current LW codebase.
All the drawcommands work is done, but I've got an issue, not directly related with that.

Cannot get a webgl context with canvas.getContext once a previous 2d context has been created on that canvas. Do you know any workaround?

tbfleming, 2017-08-03:
@jorgerobles that limit's part of the canvas spec. Only workaround is to create a new canvas.

jorgerobles, 2017-08-03:
Ok then :)

tbfleming, 2017-08-03:
Canvas has a transparent background; you can overlay one over the other when you need both 2D and 3D at the same time.

jorgerobles, 2017-08-03:
@tbfleming Ok, I've managed to get WebGL context on VideoPort. now I've got this:

[image]

https://github.com/LaserWeb/LaserWeb4/blob/camera_fx/src/lib/webcam-fx.js
Any clue?

tbfleming, 2017-08-03:
I tried to reproduce, but got this:

```
Module not found: Error: Cannot resolve module 'aruco-marker' in C:\Users\Todd\Desktop\LW\LaserWeb4\src\components
```

Is everything pushed?

jorgerobles, 2017-08-03:
aruco-marker is on package.json 😮

tbfleming, 2017-08-03:
So it is. And it's in my node_modules. odd.

tbfleming, 2017-08-03:
webpack's acting very bizarre with this:

```
C:\Users\Todd\Desktop\LW\LaserWeb4\node_modules\aruco-marker doesn't exist
```

but this works:

```
C:\Users\Todd>dir C:\Users\Todd\Desktop\LW\LaserWeb4\node_modules\aruco-marker
 Volume in drive C has no label.
 Volume Serial Number is 02AF-E41B

 Directory of C:\Users\Todd\Desktop\LW\LaserWeb4\node_modules\aruco-marker

08/03/2017  08:37 AM    <DIR>          .
08/03/2017  08:37 AM    <DIR>          ..
08/03/2017  08:37 AM                12 .npmignore
08/03/2017  08:37 AM                 5 .tern-port
08/03/2017  08:37 AM                81 .travis.yml
08/03/2017  08:37 AM             2,731 aruco-marker.js
08/03/2017  08:37 AM               270 bower.json
08/03/2017  08:37 AM    <DIR>          build
08/03/2017  08:37 AM               157 CHANGELOG.md
08/03/2017  08:37 AM    <DIR>          demos
08/03/2017  08:37 AM             1,951 Gruntfile.js
08/03/2017  08:37 AM             1,059 MIT-LICENSE.txt
08/03/2017  08:37 AM               991 ng-aruco-marker.js
08/03/2017  08:37 AM             2,553 package.json
08/03/2017  08:37 AM             2,987 README.md
08/03/2017  08:37 AM    <DIR>          test
              11 File(s)         12,797 bytes
```

jorgerobles, 2017-08-03:
O_o  
Maybe has to be related with the use of require for instancing :|
`export const ArMarker=require('aruco-marker');`

tbfleming, 2017-08-03:
false alarm. An old server process was still running on my machine; they don't always die when I exit my editor.

tbfleming, 2017-08-03:
Here's what's happening:
* first time through: canvas.getContext, create DrawCommands, dispose DrawCommands
  * dispose DrawCommands disables the gl context to free all its resources
* second time through: canvas.getContext
  * oops. That context is already disabled

You need to hold onto DrawCommands as long as the canvas exists. You need to dispose it when the canvas goes away (hot reload or tab change).

jorgerobles, 2017-08-05:
@tbfleming Could Please, Could We have an alpha parameter on `image` drawCommand

tbfleming, 2017-08-05:
@jorgerobles done

jorgerobles, 2017-08-05:
Love it! Thanks

jorgerobles, 2017-08-05:
Oye

tbfleming, 2017-08-05:
[image]

jorgerobles, 2017-08-05:
Hahah!

domenic-d, 2018-01-22:
Closed due to this being a very specific requirement for our machine and we no longer need it.

closed 2018-01-22 by domenic-d