LaserWeb/LaserWeb4 #223: viewBox bug

tbfleming, 2017-03-19:
Corel files trigger a bug in viewBox handling. I traced it down to a difference in how lw.svg-parser and LW4 code handle units.

Repo case:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.0//EN" "http://www.w3.org/TR/2001/REC-SVG-20010904/DTD/svg10.dtd">
<!-- Creator: CorelDRAW X7 -->
<svg xmlns="http://www.w3.org/2000/svg" xml:space="preserve" width="457.285mm" height="135.043mm" version="1.0" shape-rendering="geometricPrecision" text-rendering="geometricPrecision" image-rendering="optimizeQuality" fill-rule="evenodd" clip-rule="evenodd"
viewBox="0 0 38014 11224"
 xmlns:xlink="http://www.w3.org/1999/xlink">
 <g id="Layer1">
  <metadata id="CorelCorpID_0Corel-Layer"/>
  <polygon fill="none" stroke="yellow" stroke-width="6.33459" points="3,3 38011,3 38011,11221 3,11221 "/>
 </g>
</svg>
```

lw.svg-parser units question: https://github.com/lautr3k/lw.svg-parser/issues/8

lautr3k, 2017-03-20:
Strange...

tbfleming, 2017-04-09:
I traced it down to this magic number in lw.svg-parser's _normalizeTagAttrUnit:

```
        if (stringValue.indexOf('mm') !== -1) {
            return floatValue * 3.5433070869
        }
```

This is hard coding 90 px per inch while reading the ```width``` and ```height``` attrs.

tbfleming, 2017-04-09:
It uses those attributes to set up an initial matrix, which then throws of the units in the entire file.

jorgerobles, 2017-04-09:
Thats a typographic dot measure 🙄

jorgerobles, 2017-04-09:
Maybe related with Text parsing?

tbfleming, 2017-04-09:
It mangles the units of all paths. I haven't finished tracing, but I suspect it's messing up other things besides paths also.

jorgerobles, 2017-04-09:
Is probably legacy of a graphic design software tweak (AI, inkscape?)

tbfleming, 2017-04-09:
90 is Inkscape's old px per inch value. 90 px per inch = 3.5433070866 px per mm (rounding error in the hard-coded value above).

jorgerobles, 2017-04-09:
ok, I missguess (https://www.google.es/webhp?sourceid=chrome-instant&ion=1&espv=2&ie=UTF-8#safe=off&q=typography+point+to+mm)

So that value will be better to be passed as parameter isn't it?

tbfleming, 2017-04-09:
> So that value will be better to be passed as parameter isn't it?

Short term, yes. Long term it'd be nice to autodetect px per inch. The parser tries to do that but fails:
* It uses width, height, viewbox to figure out the units (ok)
* It assumes 90 px per inch when reading width (not ok, throws off above calculation)

tbfleming, 2017-04-09:
Actually, best short term: comment out these lines in _parseViewBoxAttr()

```
        this.tag.scale(scaleX, scaleY)
        this.tag.translate(-translateX, -translateY)
```

This is incorrectly trying to do a transform that we don't expect anyway. My code fixes up the px per inch and takes care of the viewbox.

jorgerobles, 2017-04-09:
We have a fork of svg-parser. Lately I've just copied src files to lib/vendors, as npm use of git repo drives the code nuts on building (I'm sure I'm doing bad, but have no explanation of what is doing that behavior)

tbfleming, 2017-04-09:
I think I'm going to drop the npm module and bring it in as a git submodule instead. That should make maintenance easier.

tbfleming, 2017-04-09:
I switched it to a git submodule and bypassed the bad xform. This can use a lot of testing. Hopefully it helps with all the px-per-inch and misaligned embedded image issues.

jorgerobles, 2017-04-16:
Closing for now.

closed 2017-04-16 by jorgerobles