LaserWeb/LaserWeb4 #225: LW4 Electron exposes old LW3 test interface

cprezzi, 2017-03-20:
When startting the electron exe, you could open the testenvironment (LW3) in a separate browser on localhost:8000. This can confuse users.

This environment was used for testing during development of lw.comm-server. It's planned to convert it to a slim LW4 remote (for mobiles & tablets).

openhardwarecoza, 2017-03-20:
That comes from your /app in lw.comms-server - can we remove it from there?
Once gone I can change the path of the lw4 copy from /app/lw4 to /app

(/app is a hardcoded requirement of electron-builder, coincidence to your
selected path for the testing)

cprezzi, 2017-03-20:
We should move the testapp (which will be converted into mobile remote app) in a subfolder like /app/mobile and have LW4 it the /app folder.

openhardwarecoza, 2017-03-20:
That'd work too (:

I am already at work, would you have a moment to fix that in
lw.comms-server repo? (i'll merge back down to electron_bundler branch)

cprezzi, 2017-03-20:
@openhardwarecoza I've swaped the webpages in the **electron branch** (moved testpage to /app/mobile and /app/lw4 to /app & changed package.json and server.js accordingly). I was able to test the electron build for Win x64. Not tested for the rest. :)

cprezzi, 2017-03-20:
Version is now 4.0.45

openhardwarecoza, 2017-03-20:
Cool! Will run a set tonight (:

openhardwarecoza, 2017-03-22:
New binaries are out

closed 2017-03-22 by openhardwarecoza