LaserWeb/LaserWeb4 #226: Replace document

DarklyLabs, 2017-03-22:
We regularly find ourselves needing to alter a document externally after applying operations and running the job.

Sometimes we have many operations applied to a document, especially if stroke filtering is used.

Being able to replace an existing document which has operations applied to it, and not have to re-create the operations would be a big time saver.

jorgerobles, 2017-03-22:
True, I found myself doing the same, but have no easy way to do this on UI.
By example, I had opened a document, and deleted some vectors for whatever reason. If you replace it, would come back unmatched vectors or other conditions that have to solve.

tbfleming, 2017-03-22:
I recommend this flow for cases like this:
* Use color filtering only; no manual separation into operations
* Delete the old document from the top. This will strip it from the operations
* load new document
* drag the whole doc into each op. The color filters will do the rest.

DarklyLabs, 2017-03-22:
Thank you for the work around.
@openhardwarecoza Can I assume this is as far as this feature will be considered and I should close it?

jorgerobles, 2017-03-23:
The most We can currently do on this, is to replace the document references upon replacing.
But that will strip unmatched inner paths if were used on operations... thus confusing, I think.

The other side issue is how to integrate replacing on current UI. That will require a context menu on the document root, or another open button each document loaded, or figuring out keyboard and click combination (Shift+Click) on some place.

DarklyLabs, 2017-03-23:
We worked with Todd's workaround today. Although not the ideal solution, it did work and saved us more time that having to re-create the operations from scratch.

jorgerobles, 2017-03-24:
@DarklyLabs Replace is available opening document with shift-click

closed 2017-03-24 by jorgerobles

DarklyLabs, 2017-03-24:
@jorgerobles Works well. Thank you.
The only 'save' we found not to work was the one on MachineProfile.

jorgerobles, 2017-03-25:
> The only 'save' we found not to work was the one on MachineProfile.
@DarklyLabs, sorry, what one?

DarklyLabs, 2017-03-26:
[image]

jorgerobles, 2017-03-27:
@DarklyLabs thanks, but that is not supposed to open a dialog, but updates the current configuration over the selected machine, and also does not work on contributed (*id) definitions. Was added as requested sometime ago. I should relabel the button. Any suggestions?

DarklyLabs, 2017-03-27:
Since the ability to save/load profiles is offered in the 'Tools' section, perhaps re think think this area and remove the 'save' & 'delete', leaving only the 'apply'.

This will need 'delete' to be available on the tools section. 

Just a suggestion but will simplify for users as this is a rarely used feature.

cprezzi, 2017-03-28:
I would keep the Save button where it is, just deactivate it when the actual profile is a *id.

cprezzi, 2017-03-28:
Hiding the save/delete options in the tools section is too far away. Usually tools is not touched.

jorgerobles, 2017-03-28:
What if rename "save" to "sync?"?

DarklyLabs, 2017-03-28:
@cprezzi I guess your opinion and decision is how it will end up. Thank you for considering our suggestion.

cprezzi, 2017-03-28:
@jorgerobles I think "save" is clearer than "sync".

@DarklyLabs I think you are right that it can be confusing, because we mix up fixed profiles with user changeable profiles in the same dropdown, but I don't see an elegant solution yet to make it obvious. 
For people with multiple machines (like me) it is very usefull to have the profile dropdown right on top of the settings and not in a hidden subtab, but probably somebody has a good idea for a more elegant solution.

jorgerobles, 2017-03-28:
Finally used "Update" label. Straight :). Also disabled the Delete/Update to hardcoded profiles, and added title definitions of use.

jorgerobles, 2017-03-28:
@DarklyLabs  I've improved the document tree UI, now when you select an object trails back their parents, light blue. That is an important visual cue.

DarklyLabs, 2017-03-28:
@jorgerobles Looks good. Thank you.

commit f9135d9c (2017-03-24): Fix #226. Implemented replace