LaserWeb/LaserWeb4 #228: Hidden vectors are calculated in gcode

DarklyLabs, 2017-03-22:
Vectors which have been hidden are still taken into account in gcode generation
Why?

jorgerobles, 2017-03-22:
Actually hidden is not deleted. If you have thrown them into operation should be used. 
It's easy to filter them.  @openhardwarecoza what do you think?

jorgerobles, 2017-03-22:
Thinking again, that would be a mess. If you forget to unhide some deep tree path all operations would be affected, and Tree navigation is not easy if you have a lot of paths. Its preferable to use color filters, as they are on scope, and more predictable to do #229

DarklyLabs, 2017-03-22:
I think this bears thinking about as the hiding feature is not able to be fully utilised easily with large jobs. It's ok for simple testing but in reality jobs can become quite complex.

If hiding was to be really useful, then the ability to 'unhide all' or 'hide all' etc would be useful.

jorgerobles, 2017-03-23:
Actually if you hide upper tree branches it propagates to inner paths. So unhide/hide all is somewhat supported. But that does not skip user frustration when forgot to unhide that tiny window on the victorian house he made.

openhardwarecoza, 2017-03-23:
In lw3 we rendered hidden paths as black lines. Then it was always visible
when you worked with it. In the half finished tree cam of lw3, you could
have deleted or not selected the hidden paths. Or used them if you wish

jorgerobles, 2017-03-23:
Taking a vague reference of Artcam, it makes the same way, that is, layers
(documents) visibility does not relate to the toolpaths (operations)

> Actually if you hide upper tree branches it propagates to inner paths. So
> unhide/hide all is somewhat supported. But that does not skip user
> frustration when forgot to unhide that tiny window on the victorian house
> he made.

AHVr28c35jMXj4NDx6hwNsWZaJHQJDGXks5roh9fgaJpZM4Mk0av>

closed 2017-03-25 by jorgerobles