LaserWeb/LaserWeb4 #230: Ability to move gcode in workspace

DarklyLabs, 2017-03-22:
Many users run repetitive jobs by loading in a gcode file. It's faster and more failsafe once they have a project that is working perfectly for them.

Having the ability to reposition gcode in the workspace in a natural extension of allowing users to work with gcode files directly.

jorgerobles, 2017-03-22:
That will require a huge preprocessor depending the generated gcode (Gcode is rather verbose). Reprap software (as far as I know) does not do that either.

jorgerobles, 2017-03-22:
@cprezzi had some G+ answers about a similar topic.

DarklyLabs, 2017-03-22:
@jorgerobles Where are these answers?

jorgerobles, 2017-03-22:
Sorry https://plus.google.com/113059362277072411228/posts/KDTmzSbbrMi. Basically It will throw LW on its knees if file is large.

DarklyLabs, 2017-03-22:
@jorgerobles  If too difficult to implement then we understand.
In effect, we are not suggesting editing the gcode, but rather transforming it. I suspect you will tell me this constitutes the same thing!

jorgerobles, 2017-03-22:
Yes, it is 🙁  .The problem is handling the amount of data. Currently load once and then is cached. Transforms do that every frame during movement.

openhardwarecoza, 2017-03-22:
Teach user to export workspace instead of gcode?

DarklyLabs, 2017-03-22:
Understood.
Thanks for the consideration.
Closing now.

closed 2017-03-22 by DarklyLabs

tbfleming, 2017-06-18:
Update: there's now a partial solution to this using the new MPos/WPos system with Set Zero.