LaserWeb/LaserWeb4 #231: UI Enhancement: Unreadable Inputs

openhardwarecoza, 2017-03-22:
[image]

tbfleming, 2017-03-22:
I love that tag!

cprezzi, 2017-03-22:
The same problem exists in the cam tab.
Could we just make the default width of the sidebar a bit wider?

tbfleming, 2017-03-22:
I think the cam tab's issues are related to side-by-side fields and other new things that were added since I designed the original CSS.

cprezzi, 2017-03-22:
@tbfleming For example: It happens that the red cross for deleting an operation is out of view with default width.

jorgerobles, 2017-03-22:
Easy solution, store splitter position on localtorage ;)

tbfleming, 2017-03-22:
I think the red cross started its disappearing act around the same time the "hide" link was added. The link forcibly widened the inner column, causing everything inside to readjust to match, flowing past the the outer column's width.

tbfleming, 2017-03-22:
> Easy solution, store splitter position on localtorage

That will help. Long term I need to go back and rework the CSS to support the added controls better.

jorgerobles, 2017-03-23:
@tbfleming Yes, will have to think deep around UI and UX. Bootstrap has some limitations also.
@openhardwarecoza does splitter remembering help with that inputs?

openhardwarecoza, 2017-03-23:
I'll try and work on the CSS for the Jog tab today first (:

openhardwarecoza, 2017-04-08:
[image]

Fixed the major issues - Not done yet - but will continue in the morning.  2:45 AM this side and brain is freezing

closed 2017-04-08 by openhardwarecoza

cprezzi, 2017-04-08:
Love that colors :)

openhardwarecoza, 2017-04-08:
Also from the testing group - those not familiar with cartesian system, the colors help give some visual clue of associating viewer with buttons, and now also DRO

[image]

jorgerobles, 2017-04-08:
Niice!