LaserWeb/LaserWeb4 #232: Settings Tab - cannot click any headers after Camera

openhardwarecoza, 2017-03-22:
Havent looked but suspect missing closing tag in caamera section.  The headers of the panels macros and tools, are not clickable anymore and cannot be expanded

jorgerobles, 2017-03-22:
Hmm.. I've been pushing some changes just half an hour. Please tell me the steps to reproduce. I've got it running, and cannot find the error

openhardwarecoza, 2017-03-22:
I dont have the half hour ago code... let me rebuild some new binaries.  This was with Last Night's binary

jorgerobles, 2017-03-22:
ok. then was my fault. I've just fixed a bug with floating camera.

openhardwarecoza, 2017-03-22:
Cool. then its already sorted.  If it works there it will work when the new binaries are done (using git to store the 57mb builds is not ideal.  I build on windows, push to git.  Boot the linux vm, then I first have to git pull the windows binaries down, before I can push the linux builds.  Those "irrelevant platform" pulls waste about 240mb per day (more if there was mac builds) and sometimes if theres a blunder, two or three times I have to go through it.   That all at average of 40kbps, building binaries became a 2 hour job (;

Considering using cncpro.co instead - one direction only: Upload

jonbev, 2017-03-22:
Is it worth looking at travis-ci to do the builds automatically?

openhardwarecoza, 2017-03-22:
Probably. But i (and most of the other devs) are quite stretched for time.
If you feel like trying? Would be great?!

The only possible pitfall is that the native dependencies need to compile
on the actual platform. Cant build for mac and windows on a linux host for
example. Dont know how the CI services handle that

jonbev, 2017-03-22:
I'd happily give it a try.

Do you have build scripts somewhere?

openhardwarecoza, 2017-03-22:
@jorgerobles  - fixed in newest build! Closing

@jonbev  - lets move this into a new issue. gimme 15 and i'll put together the whole story for you (:

closed 2017-03-22 by openhardwarecoza