LaserWeb/LaserWeb4 #233: Redundant workspace save dialog box

DarklyLabs, 2017-03-23:
Saving the cam workspace first opens up a filename dialog box, then after hitting 'ok' the file dialogue window opens.

The first filename window is redundant as the following dialog window allows the filename to be set.

tbfleming, 2017-03-23:
At one point we decided against this double dialog: https://github.com/LaserWeb/LaserWeb4/issues/93#issuecomment-268041970 . I don't remember when this was reversed.

jorgerobles, 2017-03-23:
@DarklyLabs, @tbfleming 
That's ok. I forgot #93. Anyway, I found that other pieces of code used prompt.
Not a problem, can be stripped. Following #93, most browser users does not choose to save as as the preference and goes al thru, getting "gcode 20.gcode" some days.

Last decision, should we strip all saving prompts on Electron builds, Browser or both?

DarklyLabs, 2017-03-23:
Having the 'file browser' come up for a save or load operation in all versions would be ideal, would it not?

jorgerobles, 2017-03-23:
Native file browser is not available to javascript on regular browser. Is a private chrome setting: :|

tbfleming, 2017-03-23:
We could point people to that setting when they complain about numbered files.

closed 2017-03-24 by jorgerobles

jorgerobles, 2017-03-24:
Prompt is disabled by default. Prompt Prone People should do click-shift on save

tbfleming, 2017-03-24:
Prompt Prone People Prefer Prescribed Preparations while Pondering Place

commit cdc49415 (2017-03-24): Fix #233 Stripped prompt from all saving.
commit cc6d6566 (2017-03-24): Fix #233 Trigger prompt if shift-click