LaserWeb/LaserWeb4 #234: Can I cause multiple passes to happen in sequence with the operation list?

PeterGrace, 2017-03-24:
Sorry if this is answered in the docs, I did not stumble upon it while I was looking.

I'm trying to do a job that involves an engraving, 2-pass hole cuts and finally 2-pass part cuts.  I specified multiple passes for both the hole cuts as well as the part cuts, but the gcode generator created a toolpath as such:
engrave (pass 1/1)
cut (pass 1/2)
part (pass 1/2)
cut (pass 2/2)
part (pass 2/2)

The downside to this is that if the part manages to cut through when the hole is not fully cut, the part is ruined.  Is there a way that I can cause the gcode generator to instead produce the following toolpath:
engrave (pass 1/1)
cut (pass 1/2)
cut (pass 2/2)
part (pass 1/2)
part (pass 2/2)

With this method, I can be certain that the "cut" pieces have fallen out appropriately and pause, if necessary, to run additional cut passes prior to it beginning the final part burn.

Thanks in advance for any feedback you can provide!

tbfleming, 2017-03-24:
@jorgerobles it used to run all passes for an operation before moving to the next.

jorgerobles, 2017-03-24:
@PeterGrace is compulsory you attach the workspace.json file in order to check this issue. Sorry to bother you.

@openhardwarecoza Can we put that answer templates on github, plz?

@tbfleming Will check again upon attach, but this topic have arised in at least two other issues and were well checked, including a full rehaul to follow operations instead of grouping engravings as was before (https://github.com/LaserWeb/LaserWeb4/commit/4aeefb6b90be849df8f24e90ad5c4efb061858ca)
Please check if I slipped something, thx.

jorgerobles, 2017-03-24:
@PeterGrace answering your question, the workflow is to separate on to operations. Topmost engraving, then inner cuts, then outer cuts. please check this bundle:

[platform-workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/867184/platform-workspace.json.txt)
[platform.gcode.txt](https://github.com/LaserWeb/LaserWeb4/files/867186/platform.gcode.txt)
[platform.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/867185/platform.svg.txt)

tbfleming, 2017-03-24:
Sorry @jorgerobles I jumped the gun on that one. His description sounded like he created 3 operations and that it wasn't ordering passes correctly; your example does order them correctly. Now I'm suspecting he only created 2 ops. @PeterGrace please attach your workspace.

PeterGrace, 2017-03-24:
You are most certainly right; I reloaded the workspace this morning to double check and it was indeed two operations and not three.  Apologies for being dense, and thank you for being so proactive in your response.

closed 2017-03-24 by PeterGrace