LaserWeb/LaserWeb4 #235: Cannot open raster file

lautr3k, 2017-03-25:
- With a blank workspace.
- Add an image as document.

Console message:
```
error: TypeError: Cannot read property 'naturalWidth' of undefined
    at processImage (document.js:151)
    at loadImage (document.js:196)
    at Object.reducer [as default] (index.js:72)
    at dispatch (createStore.js:179)
    at persistStateMiddleware.js:26
    at action (index.js:30)
    at index.js:103
    at promisedImage.then.img (cam.js:223)
```

jorgerobles, 2017-03-25:
@lautr3k  I'm away right now, bug is on https://github.com/LaserWeb/LaserWeb4/blob/f9135d9cece04cedee2bc943e5b1882406c21f99/src/reducers/document.js#L149

Substitute {file,context} to context, plz

openhardwarecoza, 2017-03-25:
> @lautr3k <https://github.com/lautr3k> I'm away right now, bug is on
> https://github.com/LaserWeb/LaserWeb4/blob/f9135d9cece04cedee2bc943e5b188
> 2406c21f99/src/reducers/document.js#L149
>
> Substitute {file,context} to context, plz

I am also pretty much offline till the 4th of april. @lautr3k you have push
access (: would you mind fixing?

commit 6d417f89 (2017-03-25): Fix #235
closed 2017-03-25 by lautr3k