LaserWeb/LaserWeb4 #236: Weird Image displacement.

jorgerobles, 2017-03-25:
Images are displaced upon loading. Also thrown a https://sourceforge.net/p/jsclipper/tickets/18/ error

Workspace Bundle
[container-02.zip](https://github.com/LaserWeb/LaserWeb4/files/870224/container-02.zip)

jorgerobles, 2017-03-25:
@tbfleming, @lautr3k

tbfleming, 2017-03-25:
Three bugs in one test case!
1. jsClipper exception: this one causes it to abandon triangulation (fill support). It falls back to displaying outline only.
2. The viewBox units bug (#223)
3. Offsetting: looks like it may be https://github.com/lautr3k/lw.svg-parser/issues/7

tbfleming, 2017-04-09:
I just fixed lautr3k/lw.svg-parser#7 . Try this now.

jorgerobles, 2017-04-10:
Partial success!

**Before**
[image]

**After Inkscape**
[image]

**After AI**
[image]

Images are gone on render (Js log throws some Infinity problem on Floating Window. )

Attached svg
[container-02.svg.zip](https://github.com/LaserWeb/LaserWeb4/files/909276/container-02.svg.zip)

jorgerobles, 2017-04-11:
@tbfleming images went with width and height null. Any fixups?

jorgerobles, 2017-04-11:
@tbfleming found some infinitys out there (those expensive cars)
[image]

jorgerobles, 2017-04-11:
I think the problem is instancing the image on addChildren. width is not met always. I used a promise on the image document load, but I think cannot use it here :/

tbfleming, 2017-04-11:
Don't set i.width and i.height; it defeats getting the image's native width and height. They're not null when I run it; is this another OSXism?

jorgerobles, 2017-04-11:
Can I use rawW and rawH? I found that transforms behaves wrong with globalMatrix. I've switched to localMatrix and works like a charm

jorgerobles, 2017-04-11:
Cannot find the use of `let i = new Image;` anyway I think is not needed even.

jorgerobles, 2017-04-11:
I've done some tests and the patch introduced works reasonably well :)

[image]

tbfleming, 2017-04-11:
[stars.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/914715/stars.svg.txt)

Inkscape:
[image]

old version: 93159b2:

[image]

new version: 7744f9a:
[image]

jorgerobles, 2017-04-12:
@tbfleming then we have an AI vs Inkscape war :)

Is not that bad. Seems something to do with the image proportions, but anchoring is ok. Will try on other machine to replicate the issue if has to be related with the Infinity issue.

tbfleming, 2017-04-12:
If it's Inkscape vs AI, then Inkscape should win since it's open source. It might not be that. Try constructing a simple test case in AI that mixes a small number of paths and stretched bitmaps. No groups or layers. The test needs to make it easy to see if alignment and stretch are handled correctly.

FabCreator, 2017-04-12:
AI is important to support, while its not open source it is one of the main programs used for designing. it would be a shame for laserweb to not support mainstream apps :(

tbfleming, 2017-04-12:
I doubt it would be one vs the other. Let's start with a simple case to see.

jorgerobles, 2017-04-12:
@tbfleming I think is not the case (aforementioned war)
Yesterday I made all the tests with this file, coincidentally the specs you say.

[image.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/915812/image.svg.txt)

jorgerobles, 2017-04-12:
I'm more leaned towards image instancing does not have instant measurements, as have done earlier with other code to get naturalWidth, etc I used image onload promise.

tbfleming, 2017-04-12:
Dang. Not good.

old:
[image]

new:
[image]

Which means: old code preserves Inkscape alignment and new code preserves AI alignment.

FabCreator, 2017-04-12:
could we have an option to enable inskape or ai?

jorgerobles, 2017-04-12:
> Don't set i.width and i.height; it defeats getting the image's native width and height. They're not null when I run it; is this another OSXism?

@tbfleming checked on linux also... linuxism :smile:

cprezzi, 2017-04-12:
Is there any way we can detect from the file, if it was made by Inkscape or AI?

jorgerobles, 2017-04-12:
Gonna fiddle a bit, maybe is not necessary, anyway both software use some
identificative tags, easy to detect

jorgerobles, 2017-04-12:
@tbfleming found! as I've suspected naturalWidth and naturalHeight are not available on instant of decoding image in a predictable way. Sometimes is, sometimes not. Hence, scale is wrong as should be calculated with `rawW/i.naturalWidth` for example

I was looking the place to use the promise to get those values loaded, but cannot find a suitable one. I suppose the best place should be the parser instead the reducer. Can you help me to find it?

jorgerobles, 2017-04-12:
A possible solution is to install https://github.com/gaearon/redux-thunk that allows promised actions, so documentsLoad will be a promise and dispatch upon processing.

tbfleming, 2017-04-12:
@jorgerobles there's callback and promise code in the cam component's file loading. You can add more async stuff there.

jorgerobles, 2017-04-12:
Yes I know :), then move documentsLoad code from reducer to cam and then use documentLoad as a simple state merge?

tbfleming, 2017-04-12:
That will work. Another option is for the async code to use the images to set our own attributes on Tag. The reducer would read the new Tag attributes instead of loading the images.

jorgerobles, 2017-04-12:
@tbfleming Fixed!! :tada::tada::tada::tada::tada:

closed 2017-04-12 by jorgerobles

cojarbi, 2017-04-12:
So now I plain English what's was fixed? :)

FabCreator, 2017-04-12:
somthing to do with SVGs not loading in the right place and it being different for inkscape and ai

cojarbi, 2017-04-13:
I this fix is something we need to communicate to the users , if so what should be described

jorgerobles, 2017-04-13:
So...
New version fixes embedded images bug on svg, and better support to design
software files ;)

tbfleming, 2017-04-13:
There are multiple fixes in this and related issues:
* Some paths got messed up: https://github.com/lautr3k/lw.svg-parser/issues/7
* SVG files that were anything other than 90 px per inch got messed up, even when the user set the px per inch setting correctly
* Embedded images sometimes got mispositioned or missized

tbfleming, 2017-04-13:
Corel: chances are Corel is still problematic. From the files I've seen in the wild, it looks like Corel randomly generates a different px-per-inch scale for each individual file. We can't handle that until we add px-per-inch auto detection.