LaserWeb/LaserWeb4 #238: How to run with webcam support?

punkoffice, 2017-03-28:
Hello, I'm testing out the dev-es6 branch.  Got the live environment up and running with no error messages in the console, however my webcam does not appear in the settings drop-down list.  I've run the binary and my webcam appears and works, so I'm thinking there's something I need to configure for the live environment to see it.  This is on a Windows computer.  Is there something I might have missed?

jorgerobles, 2017-03-29:
Hi! Dev-Es6 is the same as the binary. Be sure the testing url. 
Camera is available at http://localhost:8080, not http://localhost:8080/webpack-dev-server/

Further checks:
- Camera access depend on browser permissions. If you have disabled permissions to access the camera will not work.
- Upon starting, LW scans for available video devices. The log could be found in the developer tools pane (Chrome Dev tools). That should tell you a hint what is going on.
- Please state model of your camera.

punkoffice, 2017-03-29:
Thank you Jorge.  That helped.  I saw that my webcam had no label (from the JSON string) and appeared as a thin blank line in the drop-down list just after NONE which I was able to select.  Now every time I run it it appears as "USB 2.0 Camera".  Strange, but it works.

closed 2017-03-29 by jorgerobles