LaserWeb/LaserWeb4 #240: Add keyboard jog support

cojarbi, 2017-04-02:
Could it be possible to add keyboard jog support?

cprezzi, 2017-04-03:
Should be possible, it's just a bit complicated to find keys that make sense and are free on all plattforms ;)

jorgerobles, 2017-04-03:
Highjacking thread ---I want live feedback also! That is, clicking on the workspace and live feedback enabled, the head moves to the clicked point :smile:, Is useful to aim the laser to an scrap material
@tbfleming is there a way to achieve that?

cojarbi, 2017-04-03:
Highjaker !!! I think chillippepr has this implemented , not sure if it works for us

jorgerobles, 2017-04-03:
I've tried, but got lost on CastingXRays or something mutant

tbfleming, 2017-04-03:
@jorgerobles yes, but all 3 mouse buttons are currently used. WorkspaceContent.xyInterceptFromPoint converts (e.pageX, e.pageY) to machine X, Y.

jorgerobles, 2017-04-03:
Then using shift+click?

tbfleming, 2017-04-03:
shift+click is safe as long as the mouse isn't over a document.

tbfleming, 2017-04-03:
[image]

jorgerobles, 2017-04-03:
Ok, will find some other key combo ;)

tbfleming, 2017-04-03:
shift+ctrl+click and alt-click are safe. If you're really ambitious you could chord the letter keys (e.g. Q+W+shift), although that might not work on OSX and windows users only expect that type of chording on video games :)

jorgerobles, 2017-04-03:
LOL. there's a package already installed that I think it does. :D

cojarbi, 2017-04-03:
A+S+D+F is more comfortable if we want to use all fingers

jorgerobles, 2017-04-03:
Asdf or asdw?

cojarbi, 2017-04-03:
Asdw would give me the finger lol

tbfleming, 2017-04-03:
My hand rests easily on Q+W+E+R+C. It kind of reminds me of a Tetris piece.

cojarbi, 2017-04-03:
Real question: isadding a key  a safety feature or a technical one? UGS for example does not add a key but you need to check the option On first

tbfleming, 2017-04-03:
I just realized I'm mixing up the two issues. @jorgerobles hijacked my mind when he hijacked the thread :)

jorgerobles, 2017-04-03:
Sorry, highjacking. But even are different topics It sparked because are
related. I bet @cojarbi was trying to move the jog to a material, isn't it?

cojarbi, 2017-04-03:
Look what you did @jorgerobles :). Now what's what?

tbfleming, 2017-04-03:
Keyboard jog support:
* gamers who place their mouse on the right side are used to WASD for motion control (mouse aims gun)
* gamers who place their mouse on the left side (me) are used to arrow keys for motion control
* If we support both, then we'll end up supporting all machines, even the ones with crippled keyboards

Click to jog:
* I'm OK with shift+ctrl+click or alt+click (or one of the bizarre keys+click on OSX)

jorgerobles, 2017-04-03:
Click to jog: I will go for alt-click. Other keys are used already for modifying selection.
Keyboard jog: I would go for Arrows (Even Alt+Arrows to be uniform). Are meant for that :smile:, supporting other keys (wasd etc) could conflict with macros and so on.

jorgerobles, 2017-04-03:
Should jogging with click or keys, be restricted on Jog Tab?

tbfleming, 2017-04-03:
So far the current tab doesn't affect what users can do in the workspace. Is there a strong reason to limit it to the Jog tab?

jorgerobles, 2017-04-03:
Well I prefer not to move my machine if I mistake alt key. I know is a corner case, so just asking.

cojarbi, 2017-04-03:
Is a click or a key required for motion control with the arrows , I would love to use my other hand for something esle

cojarbi, 2017-04-03:
I assume jog will not work if machine is executing a job

jorgerobles, 2017-04-03:
@cojarbi good point. We have no safe checks so far.

I think Alt key is required at least for click, and maybe on keyboard too. You know, cursor keys are used on writing on fields.

tbfleming, 2017-04-03:
I forgot about the fields.

jorgerobles, 2017-04-03:
Also, Jog click should check if a homing is done before, or we have many wrongly angry users

tbfleming, 2017-04-03:
Forget keyboard, mouse, and touchscreen. We need this: http://www.economist.com/news/science-and-technology/21719774-do-human-beings-need-embrace-brain-implants-stay-relevant-elon-musk-enters

jorgerobles, 2017-04-03:
LOOOL :rofl:

cojarbi, 2017-04-03:
About time. I hope I'm alive to beta test

jorgerobles, 2017-04-03:
I've got it running locally. 

Keyboard jogging is running if Jog Panel is focused, that is, anything on the panel has been clicked. If you click outside, like workspace, the keyboard jog ceases to work. This is not an intended behavior, but a `react-keydown` one. Could work further on that if proves faulty.

Click jogging works well. I've limited to the Grid coordinates. Clicking outside the grid,  clamps to the max/min coordinate. I have no clear idea of what to do about homing first. Do we pop a confirm the first session click if has not been homed? Do we enable explicitly on jog panel as a toggle? Or maybe raise a flag if homed and use it to allow live jogging?

I would go for a toggle on Jog properly explained. Each time is turned on, machine will home and set the flag and then click jogging will work. if turned off, will disable.

cprezzi, 2017-04-04:
I would like to switch the workspace between edit-mode and jog-mode. 
It's very tedious already to work on the jog tab and shift, zoom, rotate the workspace, because I allways shift an object insted of the workspace.
If we do that, we get a lot of keys free for usage ;)

cprezzi, 2017-04-04:
@cojarbi Jog movements during a job will be added to the end of the queue, so no problem for the job, but probably bat after that.

jorgerobles, 2017-04-04:
I'm already on it. But we have to move a lot com/jog architecture to redux state. Is the only way to have a canonical state in the whole application.

I mean, the `pause` and `playing` vars should be moved to an 'job' state so all the components have access to it.

cprezzi, 2017-04-04:
@jorgerobles Remind that we use/show work coordinates, not machine coordinates, so you could not limit jogging by the coordinates.

jorgerobles, 2017-04-04:
Is mandatory to make a homing to use live jogging

cprezzi, 2017-04-04:
Many machines (ex. diode lasers) don't have homing-switches, but still need jogging

jorgerobles, 2017-04-04:
Then they have to look a solution to replicate the screen to their machines. I've got one, and just put the head on a corner and G92 X0Y0, or then we have to query their machines for stored position. Workspace right now is not showing work coordinates :|

jorgerobles, 2017-04-04:
Live jogging is a bit leaned to homing capable machines, like k40, you know what I mean.

cprezzi, 2017-04-04:
Workspace IS showing work coordinates after the user zeroed the axis.

cprezzi, 2017-04-04:
That's a historical thing. LW allways used only work coordinates.
But I think we need to add both coordinate systems (switchable), especially since we move to milling.

jorgerobles, 2017-04-04:
O_o .. Confusing.
The grid does not reflect negative coordinates, is showing machine coordinates rendering. 
That is a pending job (I started with machine offsets to move the axis to the real work position)

cprezzi, 2017-04-04:
I think that's a lot of work that needs to be donne correctly, not ad hoc. We should support G54 to G59.2.

jorgerobles, 2017-04-04:
@cprezzi anyway, I think does not matter if are work or machine coordinates. I not supposed to click arbitrarily out the grid (speaking of live jogging) as have no reference of what are doing. 
What kind of solution do you propose?  I can strip homing first limitation but then is up to the user to care about his machine.

jorgerobles, 2017-04-04:
And rename machineWidth and machineHeight to workWidth, workHeight :smile:

cprezzi, 2017-04-04:
Back to key jogging. We need six keys for X,Y,Z, not 4 ;)

jorgerobles, 2017-04-04:
It's done :D I have not commited yet, but supports ctrl+alt+up/dn to Z

cprezzi, 2017-04-04:
Where did you implement it? In jog.js?

jorgerobles, 2017-04-04:
https://github.com/LaserWeb/LaserWeb4/commit/2b610c52a66733d20892ed3b0128b4e9da6cb224 :)

cprezzi, 2017-04-04:
That's just the live jogging in workspace, right?

jorgerobles, 2017-04-04:
Live jogging and key jogging.

cprezzi, 2017-04-04:
Ah found, jog.js was folded ;)

cprezzi, 2017-04-04:
@jorgerobles Why did you reformat so much code? That does it make very hard to see the real changes.

jorgerobles, 2017-04-04:
Sorry, was vscode autoformat. will try to isolate.

cprezzi, 2017-04-04:
Uhh, I use brackets ;) But their autoformat is even worse ;)

jorgerobles, 2017-04-04:
vscode format is ok, just had to make it on other commit, so can read clear.

I've added the [keyUp...] decorators, added LiveJogging component.
Synced isplaying and paused to jog state on mount and unmount, so can react changes on child components. Hard access to that variables does not reflect on bounded components

cprezzi, 2017-04-04:
I have tested a bit. The alt+click on wokspace work as expected, but the keybours shortcuts don't work, because alt+left and alt+right are used by the browser for previous/next url.

jorgerobles, 2017-04-04:
my fault. should did e.preventDefault. Will fix later.

[9 more replies]

closed 2017-04-07 by jorgerobles