LaserWeb/LaserWeb4 #241: Operation Gcode Hooks

jorgerobles, 2017-04-02:
Following https://plus.google.com/107613521854079979112/posts/dkyuvd9AXcn?fscid=z13xwjoodpmhhvsiv04cfhnxlo3kvvkpn0c.1491158996355402

We have opened the posibilities of having Pre/Post Operation/Pass Gcode injection.
That will allow more flexibility of current "Enable Air assist" to by example, "Pause after operation"
Then we have several ways to do it.

- @cprezzi exposed to have an open Pre/Post Gcode field at Operation level, so the users are free to customize that operation behavior.
- Actually, we have a very specific Air Assist implementation that basically does that, having the AA code on settings, and a easy toggle at Operation level.

I was thinking an amending UI as a Hook manager on Settings (or modal) that allows to customize an specific kind of behavior, with gcode and tag label.
Then a tag-like combo (react-select) that allows the user to select what tagged behavior should the operation use. This will allow a bigger range of customization without cluttering the operations UI.

jorgerobles, 2017-04-03:
@cprezzi We can also make your way, but grouping into a folded pane on operation.
Operation fields could be 'hookOperationStart', 'hookOperationEnd', 'hookPassStart', 'hookPassEnd'

cprezzi, 2017-04-03:
We could also use marcos for that. 
The user just would define macros for every "injection" he needs and on operation level, we would only need a pre-execution and a post-execution dropdown to select the macro.

jorgerobles, 2017-04-08:
@openhardwarecoza Operation hooks are available on CAM also, If you don't mind, I prefer to have macros setup back on settings :smile:

openhardwarecoza, 2017-04-08:
Macros setup is back on Settings tab 

(Moved it afterletting the OpenBuilds team have an unsupervised go at playing with LW4 with me just watching.   Opened up a lot of usability concerns - things that from a UI perspective needs some moving around.  In this case I didnt realise you want to use Macros for the hooks (sorry) so it makes kinda sense on the Settings tab.

jorgerobles, 2017-04-08:
Thanks again. I was comitting late yesterday but forgot. Coincidence.

Anyway any ui suggestions are welcomed!!

openhardwarecoza, 2017-04-08:
(: I'll be experimenting some over the weekend.   Jog tab renamed to Control tab (It does so much more than jog since it has DRO, Macros, Play/Abort etc too) for example

Sorry if its a bit disruptive (; but hey I got to get back into coding somehow (;

closed 2017-04-08 by jorgerobles