LaserWeb/LaserWeb4 #243: Purge most of dist/

tbfleming, 2017-04-03:
Until now dist/ has been just a thorn oozing obsolete code. I've avoided removing it in the past because doing that would have produced a colossal diff that would bring GitHub to its knees. Now it has turned into a full-blown infection that threw a potential contributor miles away from the real code base (#242). GitHub fixed the big diff issue so it's probably safe to nuke it now. Be careful; it includes some non-generated files which must remain.

cprezzi, 2017-04-04:
@tbfleming I don't know what you mean. On github, there are only the 3 needed files in the dist folder. All the rest in built for publishing only and not synced to github.
We do need the generated dist files for lw.comm-server (and building the electron-installer) only.

tbfleming, 2017-04-04:
Oops; it was already purged :) The malformed pull request in #242 threw me off by referencing the gh-pages branch.

closed 2017-04-04 by tbfleming