LaserWeb/LaserWeb4 #245: File dialogue bug

DarklyLabs, 2017-04-04:
Press 'add document' to open a file dialogue window.
Without selecting anything, press 'cancel'
Trying to add a document from now on is no longer possible. No dialog box is opened.

Verified on 4.0.45 Mac and Win builds

cojarbi, 2017-04-04:
Can replicate in 4.0.48. Only way out is going to another screen from the left menu and come back.

jorgerobles, 2017-04-05:
This is very tricky. Involves react overdipatching and that fileinput does not fire any event on cancel, thus cannot restore the initial state.

DarklyLabs, 2017-04-05:
@jorgerobles Perhaps one of the other developers can shed some light on how to fix this bug?

jorgerobles, 2017-04-05:
@DarklyLabs, fixed using another hack, this seems to work on Mac. Please test on Win, thanks :)

closed 2017-04-05 by jorgerobles

DarklyLabs, 2017-04-05:
@jorgerobles Can confirm it is working correctly now in win x64 exe.
Well done.

cprezzi, 2017-04-06:
@jorgerobles The same bug still exists on the workspace load and the gcode open buttons.

cprezzi, 2017-04-06:
@jorgerobles Sorry, false alarm. I did test with the exe, but that's a day older than your commit ;)
It works great in the dev environment.

commit 8660f0f6 (2017-04-05): Fix #245
commit 182863d8 (2017-04-05): Fix #245 seems to work so far.
commit 112ac33b (2017-04-05): Fix #245 seems to work so far.