LaserWeb/LaserWeb4 #246: SVG import error

DarklyLabs, 2017-04-06:
We have been getting reports of an error that is cropping up recently when importing SVG files from multiple users.
Reports are that this is occurring with files that were previously useable in LW.

We can't replicate it in-house but it has come from multiple separate sources.
LW Version: 4.0.48 x32 & x64 (compiled 6th April 2017)

[image]

Any suggestions on the possible cause for this?

cojarbi, 2017-04-06:
Is the file loading? This was implemented a couple of weeks ago to tell the user there is something wrong with the imported file. Previously it was not shown.

DarklyLabs, 2017-04-06:
The file is not loading.

jorgerobles, 2017-04-06:
As long as file us not attached on issue cannot resolve. Thanks

DarklyLabs, 2017-04-06:
@jorgerobles Sorry, here is one of the files mentioned by our user base.
[logo_joined.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/901665/logo_joined.svg.txt)

jorgerobles, 2017-04-06:
@DarklyLabs  https://github.com/LaserWeb/LaserWeb4/wiki/Importing-SVG-from-AI

The file attached has no problems so far. It's only not using attributes, so no CSS is not parsed. @lautr3k was working on it a while ago, I don't know the status of CSS parsing :|

DarklyLabs, 2017-04-06:
@jorgerobles We are familiar with article but it is not the problem users are experiencing. We can import this file ourselves as stated in our original post. The problem is being reported by multiple users from yesterday and today and specifically with v4.0.48.
I attached the error log showing the error which they are seeing upon load.
Is there any way to determine what that error refers to and possibly use that information to help track-down why they are experiencing this?

jorgerobles, 2017-04-06:
[image]

jorgerobles, 2017-04-06:
`naturalwidth` is related with images. It's used upon load of image to position on the settings corner.
I've got no access at this moment to all the code, so I suppose is related to the fact the images are not embedded but linked, so cannot resolve the file.  

Recent code has not anything to do with that feature implemented a month ago, so I suspect it just arised now.

Please ask if are they embedding the images (that was why I linked the wiki :smile:)

DarklyLabs, 2017-04-06:
There are no images in that file. It is one we create and use within a tutorial for customers. They download it directly from our server.
I will try to find out any more information I can.

tbfleming, 2017-04-06:
My turn to hijack a thread. Maybe we should hide the "Skip tag" messages. It keeps sending people down the wrong path. e.g. https://plus.google.com/118051184495908064866/posts/c2kW42jrbxn

cojarbi, 2017-04-06:
The idea I think it was to show the user there could be an issue with the file.

DarklyLabs I tested you file in the same build on Windows and showed the same as  Jorges

openhardwarecoza, 2017-04-06:
> 118051184495908064866/posts/c2kW42jrbxn

Yip, and all errors need to be caught, and handled, not just displayed as
rephrased console.log technical message.

Catch it, then give the user a useful error message explaining what
happened.  Really technical users can go dig in the dev tools for more
details, the regular users should never see technical messages

Examples from LW3 days:

https://github.com/LaserWeb/LaserWeb3/blob/90ebc88c40da29e803f79e26e32b2863cf550667/public/js/threegcode.js#L18

https://github.com/LaserWeb/LaserWeb3/blob/8f159425428a69c089b2f137a1f10bf25a8efa9a/public/js/laserraster.js#L86

https://github.com/LaserWeb/LaserWeb3/blob/90ebc88c40da29e803f79e26e32b2863cf550667/public/lib/svgreader/SVGReader.js#L754

jorgerobles, 2017-04-06:
> 63cf550667/public/js/threegcode.js#L18
> f25a8efa9a/public/js/laserraster.js#L86
> 63cf550667/public/lib/svgreader/SVGReader.js#L754

Warns are thrown by svg parser. Was piped to devtools but after Ariel
suggestions piped to commandhistory. Is true that most of the warns are not
needed but fixed. @lautr3k seem missing lately.

openhardwarecoza, 2017-04-06:
> 90ebc88c40da29e803f79e26e32b28
> 8f159425428a69c089b2f137a1f10b
> issuecomment-292204213

(; fork man fork.

jorgerobles, 2017-04-06:
I'm on it :D

jorgerobles, 2017-04-07:
Pushed https://github.com/LaserWeb/LaserWeb4/commit/378fa70d8db2ad4a574b208089a549ffb87a95de Should skip image processing if no image (probably by a Svg parser false true)

closed 2017-04-07 by jorgerobles