LaserWeb/LaserWeb4 #247: Operation skipping parts of documents

iceblu3710, 2017-04-06:
Anybody getting the laser operation skipping parts of documents? (SVG & DXF)

It's weird, I select the whole document and choose a cut operation and some things are missing. I select them and make a separate cut operation and they produce gcode.

I have one file in particular that is a sketchup export and it appears nothing is a closed path so all the inner/outer operations do nothing. Wouldn't be the end of the world but cut code is flakey too.

[Laserweb-Workspace.txt](https://github.com/LaserWeb/LaserWeb4/files/901415/Laserweb-Workspace.txt)
[Head_Bracketsuntitled3-svg.txt](https://github.com/LaserWeb/LaserWeb4/files/901419/Head_Bracketsuntitled3-svg.txt)
[Head_Bracketsuntitled3-dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/901420/Head_Bracketsuntitled3-dxf.txt)

jorgerobles, 2017-04-08:
@iceblu3710 how did you convert to svg from dxf? Opened with AI and show a ton of duplicated overlapped paths. Colinear paths causes wreak on clipperjs.

tbfleming, 2017-04-08:
Yep, that file is bad. We can't blame clipperjs on this one; all polygon libraries will have issues with it. It's kind of like how 3d printing slicers need good triangle meshes. There are tools which try to automatically fix bad triangle meshes, but there results are unpredictable. So far I don't know of any opensource tools which try to automatically fix bad polygons for CNC.

iceblu3710, 2017-04-08:
New installs on a Windows 10 machine:
 - Sketchup MAKE v.17.1.174 64bit
 - Guitar List (current on website) [exported: millimeters, polylines] 

Loaded in LW and it was acting weird, Opened in DraftSight and saw everything was individual lines so exported to SVG, that acted weird as well.

My first time going from SketchUp to trying to cut. Usually I design in 2D and just assemble in my head, 3D makes life so easy...

Why does the clipperjs not work when the whole document is selected but when you choose individual paths it works fine?

tbfleming, 2017-04-08:
We use clipper's xor option. This allows it to interpret smaller polys inside larger polys as holes. When they overlap completely they cancel out.

iceblu3710, 2017-04-08:
Ahh, makes sense.

So what should I do to generate good data from the guitar list plugin? I am just now selecting the top faces of the objects instead of the whole object to see if that helps.

tbfleming, 2017-04-08:
Don't know. I learned to avoid sketchup like the plague back when I was heavy into 3d printing.

cojarbi, 2017-04-08:
Have you tried SketchUp stock dxf export? Inversion 2017 is included

iceblu3710, 2017-04-08:
Selecting only one face for export with the guitar list plugin works as expected. I think the plugin doesn't handle separating faces so well.

Sketchup Make 2017 does not have a dxf option under `File->Export->2D Graphic` Like the documentation says.

openhardwarecoza, 2017-04-09:
Guitarlist works best if you keep faces on your drawings, then select all before export

iceblu3710, 2017-04-09:
I arrange all my components on the X/Y plane and the select them all and 'right click -> explode'. Then I select just tbe top faces to cut and export. I can 'undo' the explode and I have components back.

closed 2017-04-09 by iceblu3710

cojarbi, 2017-04-09:
@iceblu3710  not sure why you dont have the stock dxf export option 
[image]