LaserWeb/LaserWeb4 #248: Parsing External Gcode (Yes I know we are still on TmpParser)

openhardwarecoza, 2017-04-07:
Loading gcode from sketchucam, renders some G1 moves in Green 

[image]

(Yes I know the parser is temporary - but doesnt seem like https://github.com/LaserWeb/lw-gcode-parser/pull/1 is making any more progress? Isnt it maybe better to just add G2/G3 to the tmp parser, and fix some of these bugs as we find them?

kaosat-dev, 2017-04-07:
No promises but I plan on reviving the work on the parser this week.

openhardwarecoza, 2017-04-07:
Aha! Excellent news!

jorgerobles, 2017-04-08:
Pushed some code based on cncjs gcode parser. Could be useful so far 
It's a GCodestimator https://github.com/LaserWeb/lw.guesstimate

openhardwarecoza, 2017-07-31:
Just nudging this old issue again.  Had a user via OB forum complaining about G2/G3 support again

openhardwarecoza, 2017-07-31:
Just to recap, i am saying - keep this parser just ```Isnt it maybe better to just add G2/G3 to the tmp parser, and fix some of these bugs as we find them?```