LaserWeb/LaserWeb4 #249: Dragging 3D viewer not proportional to viewPort, seems proportional to Window

openhardwarecoza, 2017-04-08:
Dragging in the viewer feels unnatural.  

Play with LW3 and see how well OrbitControls+ThreeJS felt in that respect 

I am taking a guess but it feels like mouse events are scaled from Window, instead of the actual Viewer's dimensions?  Thus mouse outruns actual move?

tbfleming, 2017-04-08:
Improved version:

[image]

Compare with LW3:

[image]

tbfleming, 2017-04-08:
I cheated with the LW3 comparison. See if you can spot what it is.

openhardwarecoza, 2017-04-08:
Tested, works much smoother thanks!

And negative, you got me stumped - can't see what the cheat is?

closed 2017-04-08 by openhardwarecoza

tbfleming, 2017-04-08:
Perspective is off in the LW4 capture; LW3 can't turn it off. With perspective, motion speed varies with distance to camera.

openhardwarecoza, 2017-04-08:
ahh yes!