LaserWeb/LaserWeb4 #250: Motors Off button isnt enabled on smoothie

openhardwarecoza, 2017-04-08:
@cprezzi it looks like you checked for 'lpc176*' previously, but the new server spits out 'smoothie' as firmware name?

openhardwarecoza, 2017-04-08:
https://github.com/LaserWeb/LaserWeb4/blob/a9b961500c95cac509dce79135e5178e2fdd5b0b/src/lib/lw.comm-client.js#L109

FabCreator, 2017-04-08:
kinda off topic. but related to motors off
What about an idle timer? not using the machine and the motors power down automatically?

openhardwarecoza, 2017-04-08:
Firmware topic.  Grbl has an Idle delay

cprezzi, 2017-04-08:
@openhardwarecoza That lib is not used. It was just needed for porting lw3 code. The used code is here: https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/components/com.js#L198, but the frontend doesn't care anymore for firmware differences, that's handled by lw.comm-server.
(See https://github.com/LaserWeb/lw.comm-server/blob/master/server.js#L427)

openhardwarecoza, 2017-04-08:
Noted, but the motorsOff button stays hidden (:

cprezzi, 2017-04-08:
I did't care because it's smoothie ;) 
No no, I just forgot to program it. Will check...

openhardwarecoza, 2017-04-08:
lol! 

I hear you (;

cprezzi, 2017-04-17:
Instead of having the old "Motors Off" button beside the XY-jog buttons, I belief it's better to instruct smoothie users to just add a "Motors Off" macro (with M84 in it). Isn't it?

openhardwarecoza, 2017-04-17:
Hmmm. Would be fine, but that is a standard button in almost all hosts.
Hard to do? Or other reason for not? Its just Grbl that doesn't support a
motors off cmd. Smoothie, tinyg, marlin, repetier all does.

cprezzi, 2017-04-17:
Hmm, I don't know any CNC host software that has such a feature (because they loose the origin, if motors would be deactivated). What is the use case for this function? Does it even have something to do with jogging?

It's easy to do in the "bad old" dom manipulation way, but it should be done correctly via react with a special button class with visible parameter. Not my speciallity ;)

openhardwarecoza, 2017-04-17:
(; cool, thats a good enough reason to leave it hanging for now.

Use case: Z knobs! I can fine tune faster by twisting the knob than
multiple jog operations (;

cprezzi, 2017-04-17:
Ah yes. I have a nice wireless handwheel for that ;)

jorgerobles, 2017-05-25:
@openhardwarecoza, @cprezzi, Sorry I'm a bit lost in this conversation. What is the functionality to achieve? Is it related to UI?

About Z knobs. There's a suggestion to include Joystick jogging on other issue. Is that worth to implement? Has to be on "server" part, though.

openhardwarecoza, 2017-05-25:
Yip mostly UI.  If firmware! = grbl (grbl doesnt have motors off. Tinyg and
smoothie does) then enable the currently hidden Motors Off button on
Control tab. That button should just send m84 to Smoothieware or whatever
tinyG's is as appropriate when clicked. It disables the motors, allowing
you to move machine manually (z knob for example is a metal knob you attach
to your Z leadscrew that you can turn by hand to zero your endmill much
more precisely, tactile and efficiently versus jogging. Its not
programmatic. It's physical. Cant turn it while motor is energised though,
so send motors off, walk to machine, move axis manually. Walk back to pc.
Set zero and go)

Re joystick: its an old mach3 thing and some people use it. A wireless usb
xbox360 remote makes it easy to jog while moving around machine/shop so
yeah, would be cool. I had it in lw1, there was a very simple js library
(cant recall now, will check, that used html5+chrome api to talk to the
gamepad plugged into machine where browser runs. Better than via nodejs

openhardwarecoza, 2017-05-25:
https://github.com/neogeek/gamepad.js/tree/master

On May 25, 2017 10:07 AM, wrote:

Yip mostly UI.  If firmware! = grbl (grbl doesnt have motors off. Tinyg and
smoothie does) then enable the currently hidden Motors Off button on
Control tab. That button should just send m84 to Smoothieware or whatever
tinyG's is as appropriate when clicked. It disables the motors, allowing
you to move machine manually (z knob for example is a metal knob you attach
to your Z leadscrew that you can turn by hand to zero your endmill much
more precisely, tactile and efficiently versus jogging. Its not
programmatic. It's physical. Cant turn it while motor is energised though,
so send motors off, walk to machine, move axis manually. Walk back to pc.
Set zero and go)

Re joystick: its an old mach3 thing and some people use it. A wireless usb
xbox360 remote makes it easy to jog while moving around machine/shop so
yeah, would be cool. I had it in lw1, there was a very simple js library
(cant recall now, will check, that used html5+chrome api to talk to the
gamepad plugged into machine where browser runs. Better than via nodejs

jorgerobles, 2017-05-25:
@cprezzi Maybe server can expose a method named `machineCapabilities` or so where lw.comm-server could expose things that have/not have specific machines, keeping the logic there abstracting including "if firm!=grbl". Capabilities could be like:
<pre>
{ 
   canStopMotors: true|false
   canDoAmazingThings: true|false
}
</pre>
Alternative is to use Macros, now are machine profile bound.

cprezzi, 2017-05-26:
@jorgerobles I like this future proove idea! We should start a list with all the features.

cprezzi, 2017-05-26:
I have started a list here: https://github.com/LaserWeb/lw.comm-server/wiki/Firmware-feature-list

closed 2017-07-08 by openhardwarecoza

cprezzi, 2017-07-10:
Ups, forgot that issue. The server now supports the getFeatureList command, which delivers a list of features. (see: https://github.com/LaserWeb/lw.comm-server/wiki/Firmware-feature-list)

cprezzi, 2017-07-10:
At the moment the delivered features are predefined by firmware, but I plan to query the firmware config for a more acurate list, if possible.