LaserWeb/LaserWeb4 #251: Check outline/size is firing the laser

FabCreator, 2017-04-08:
Hi Guys,

Recently i have noticed the Check outline/size is firing the laser?
This should be a G0 command?

We are using the latest build for 64x windows

On a second note, it might be a cool feature to have a check outline which we can set the power and use to cut out an engrave or test job etc.

Cheers,
Bonne

cprezzi, 2017-04-08:
"Check Size" uses G1 and the S value from "tool test power" setting, but if the laser was inactive (M5) before, it should not fire anyways. Does it also fire if you manually execute M5 before? 
Additionally, the default "tool test power" is 0, which should not fire (if min power is correctly configured in firmware).

This feature is very usefull for diode laser owners, to have the laser on while checking the size.

cprezzi, 2017-04-08:
I like the idea of a quick cut out feature!

openhardwarecoza, 2017-04-08:
Well, laser test, and check outline, probably could have different powers?
(another setting)

For example when using laser test to align mirrors on a co2 it needs enough
power to mark your tape clearly. When tracing outline to check if you are
on the spot, you really dont want to mark the material at all?

FabCreator, 2017-04-08:
Can it be changed to G0? seems odd to have what is a rapid movement G1? or is it because of somthing to do with diodes?

Just tested the M3 M5,
Seems the smoothies on boot up default is to have M3 active? I hope i can change that in the config. After a job is run its not a problem because jobs end with M5, then its fine unless i manually type m3 before outline.

We have a trickle power set on our PSU which means we Fire (ever so slighty even on 0%) the reason for this is we reduce the time to excited useful for rastering, also it ensures when you are engraving fine details the deceleration calculations in smoothie dont change the S value below the point the tube will fire.
So 0-100% on laserweb will always fire the laser :)

cprezzi, 2017-04-08:
@FabCreator I don't want to change it to G0 because as I said G1 it's needed for diode lasers, but I can add a separate 'check size power' to the settings.

FabCreator, 2017-04-08:
sorry i missed the bit when you said for diodes :) 
a power setting for it would be good, i guess in that case we can already use it as a cutting tool ;)

cprezzi, 2017-04-08:
Yes, that would be possible :)

cprezzi, 2017-04-10:
it's done (https://github.com/LaserWeb/LaserWeb4/commit/51fb059215e16c14e57209c9bf084e27c8f88502)

closed 2017-04-10 by cprezzi