LaserWeb/LaserWeb4 #252: Add Document Button: Filter extensions

openhardwarecoza, 2017-04-08:
Right now Add Document button allows users to select .gcode files for example (yes it does error afterwards, but it helps to filter) 

In LW3 we used the accept property of the file input https://github.com/LaserWeb/LaserWeb3/blob/087d60e630fe87a1a6bb14fad39a3a921dadda41/public/index.html#L902 

Not sure how to do this with the FileField React component

openhardwarecoza, 2017-04-08:
Can be extended to the Open Gcode button too (Only filter nc/tap/g/gcode/etc extensions

cprezzi, 2017-04-08:
It would be easyer for users, if they could open any supported file type with the "open document" button, and LW4 decides what to do with it. That should be possible.

openhardwarecoza, 2017-04-08:
Sort of, but without any filter it displays everything, video files, word docs, you name it (:
Would be good to limit it at least somewhat to usable extensions. User can always use the dropdown on the open dialog to select all files instead.  Use case for example:  Me with my hoarding desktop.  Opening desktop in the file open dialog shows everything, sketchups, videos, docs, you name it.  Makes me scroll more than if it only displayed dxf/svg/images

cprezzi, 2017-04-08:
Shure, a filter for the supportet file formats should be there.

jorgerobles, 2017-04-08:
@openhardwarecoza  Done!

openhardwarecoza, 2017-04-08:
Didnt test, but isnt dxf missing? Or is DXF and image/* mime type?

openhardwarecoza, 2017-04-08:
Ahh there we go!

closed 2017-04-08 by openhardwarecoza