LaserWeb/LaserWeb4 #254: Helper arrows remain when side menu closed

domenic-d, 2017-04-09:
[image]
[image]

jorgerobles, 2017-04-16:
Cannot consider it a bug. What's the scenario?

domenic-d, 2017-04-16:
@jorgerobles I'm not exactly sure what you mean by a scenario, but here it goes.
What does happen:
Step 1: The user minimises the CAM tab.
Step 2: The red helper arrow titled "Click here to begin' now points to the COMMS menu.

What should happen:
Step 1: The user minimises the CAM tab.
Step 2: The red helper arrow titled "Click here to begin' should be hidden along with the CAM tab.

If you cannot consider it a bug, then perhaps you may consider it 'undesirable' or 'misleading' behaviour?

jorgerobles, 2017-04-16:
ok, let me rephrase :)
 In which case the advice of  there's no documents to work is not desirable.? 
 My scenario is a noob user that runs laserweb and clicks on the tabs and then the panel is hidden. Has no loaded files, I find desirable to hint him where to start.

I'm not discarding the issue, just trying to empathize the situation :)

domenic-d, 2017-04-16:
It must at least point to the correct tab otherwise we are confusing the 'noob' even more. Please look carefully at my images and explanation.

cojarbi, 2017-04-17:
Maybe have a noob default option so it shows the help and an option to disable it?

jorgerobles, 2017-04-17:
@domenic-d You are right. Sorry, misread your former comment.

jorgerobles, 2017-04-17:
Patched using same notification scheme than settings. If full balloon is desirable, @tbfleming please explain the way to do with getbounds :)

closed 2017-04-17 by jorgerobles